Nuand/bladeRF #364: libbladeRF: Can't probe for already opened device when using CyUSB backend

jynik, 2015-01-21:
When a bladeRF is already opened using the Cypress backend, attempting to probe for a device will fail to report the currently opened device.

This is because we attempt to open the device to gather information about it. However, we use a global lock to ensure the device isn't used by multiple processes. This prevents us from gathering the probe information.

This is most easily observed via `bladeRF-cli`:
 - Open a device via `bladeRF-cli -i`
 - Run the `probe` command in interactive mode.

mambrus, 2015-01-22:
I think this could be the same case that I occasionally see. Use-case is different but end-result sounds very similar to me. Please advise with technical info about what's going on and I'll try to confirm.

jynik, 2015-01-23:
@mambrus This was specific to the Cypress backend.  Definitely different if you're using libusb in Linux.

jynik, 2015-01-23:
Addressed in 5d49a7b27111c3c1ac0121573d0c2b87739adbb4.

closed 2015-01-23 by jynik