Nuand/bladeRF #367: Make FPGA Autoloading fault tolerant

victoredwardocallaghan, 2015-01-25:
In https://github.com/Nuand/bladeRF/wiki/FPGA-Autoloading there are some scary "Important" remarks.

These make me a little nervous that one day I will forget to check the LED's have come on before plugging in the USB. Why can't the board send a 'ready' status on the USB interface? If the board isn't 'ready' yet and a bladerf_open() call has been made, bladerf_open() will block until the average time needed for a load to occur timeouts then returns with BLADERF_ERR_TIMEDOUT /* Board found but did not become ready in time */

I presume there is a bit set on the board that tells the board if it should read the SPI flash or not. This same bit could be used to auto adjust the 'ready' status. If no SPI read is needed then a 'ready' is issued, if a SPI read is needed then a 'ready' is issued at the end of the read.

P.S. I have not checked all the code on the board side to see exactly the implementation.
Kind Regards,
Edward.

jynik, 2015-01-25:
```
Why can't the board send a 'ready' status on the USB interface?
```

As of [FX3 firmware](http://nuand.com/fx3.php) v1.8.0 with [libbladeRF v1.0.0](https://github.com/Nuand/bladeRF/releases/tag/2014.11), this is now done in the bladerf_open() call, and almost exactly as you suggested.

I'll update the wiki to note this.

closed 2015-01-25 by jynik

victoredwardocallaghan, 2015-01-25:
Wonderful !

jynik, 2015-01-25:
Forgot to note -- if you run with the log level "verbosity" you should see something printed along the lines of "waiting for device to become ready" while the `bladerf_open()` call is blocked waiting for autoload to complete.  There's a limit to the number of retries which is based upon our existing knowledge of how long the FPGA /should/ take to load (and then some for good measure).