Nuand/bladeRF #369: Documentation fails to compile

victoredwardocallaghan, 2015-01-25:
While on tag 1.1.0 (not sure about newer) on NixOS the documentation failed to compile with lots of messages like this:

" error: 'CLI_CMD_HELPTEXT_run' undeclared here (not in a function) "

This only happens when pandoc is in PATH.

victoredwardocallaghan, 2015-01-25:
False alarm, distro specific issue.

jynik, 2015-01-25:
If you have a few moments, would you mind commenting a bit on the underlying issue, just so we can be aware of what was going on and what your resolution was? Thanks! :)

victoredwardocallaghan, 2015-01-25:
Actually I was wrong, this is defintely a issue here. I had orginally thought that perhaps it was just the shebang being a distro thing however that was not the case.

jynik, 2015-01-25:
I'm not famililiar with NixOS, and I'm not too sure if @rtucker is (he wrote the doc scripts).

I'll try to get a VM set up to reproduce soon and poke @rtucker for his thoughts.

Please do keep us posted if you have any additional observations or thoughts on the issue in the meantime.

victoredwardocallaghan, 2015-01-25:
OK I have traced it down a fair amount now:

I can reproduce on ArchLinux also.

the line, if [ $(grep -c "^--" <<< "$line") -gt 0 ] never branches to be true leaving the header empty between the guards as the loop returns nothing.

This is because the outputed cmd_help.txt does not have headers underscored. Here is a paste of the generated cmd_help.txt from pandoc's invocation http://pastebin.com/tst3Fqdq

I hope that is useful.
Kind Regards,
Edward.

jynik, 2015-01-29:
Thank you very much for digging into this and providing a pull request. I've got it staged up in the [dev-issue_369](https://github.com/Nuand/bladeRF/tree/dev-issue_369) branch.

Before merging to master I just want to get some a quick "thumbs up" on this change across a few distros.

jynik, 2015-01-30:
Merged your pull request in 39cd5de28030ad8af2777a9b6fa88da23c2f442b. Please mark this closed or confirm that that change was just an additional fix, aside from the issue described above.

jynik, 2015-02-17:
Confirmed this is still an issue, but the fix for #375 should resolve it.

jynik, 2015-02-17:
Resolved by @mambrus's patch in 89612e7dc361eeb34f65ac224bd84b3f7f2e6f3d

closed 2015-02-17 by jynik