Nuand/bladeRF #374: Several issues when cross compiling host tools and libs for windows on a linux system via MinGW

mnhauke, 2015-02-14:
##### 01 -  [Cmake] could not load file "FindCyApi"

Cmake could not load file "FindCyApi" since the file is actually named "FindCyAPI.cmake"
and linux handles all files case sensitive.

```
CMake Error at host/CMakeLists.txt:65 (include):
  include could not find load file:

    FindCyApi
```
##### 02 -  [Cmake] Unable to find pthread-win32 COPYING.LIB file when cross-copiling

License files for helper libraries are not always available in all build environments (e.g. crosscompiling with MinGW on Linux)
I understand that you should include a licencing file for the distribution but should this really be a hard error during buildtime?

```
[   30s] CMake Error at cmake/modules/FindLibPThreadsWin32.cmake:83 (message):
[   30s]   Unable to find pthread-win32 COPYING.LIB file
[   30s] Call Stack (most recent call first):
[   30s]   CMakeLists.txt:214 (find_package)
[   30s]·
[   30s]·
[   30s] -- Configuring incomplete, errors occurred!

```
My workaround was to to comment out some parts from windows-only post-build section in hosts/CMakeLists.txt
[buildlog](https://blackhole.pmtu.de/.dump/bladeRF-mingw32-cmake)

##### 03 -  compile error
```
[   18s] /home/abuild/rpmbuild/BUILD/bladeRF-2015.01.rc2.git1422741959.4e78b05/host/libraries/libbladeRF/src/file_ops.c: In function 'get_install_dir':
[   18s] /home/abuild/rpmbuild/BUILD/bladeRF-2015.01.rc2.git1422741959.4e78b05/host/libraries/libbladeRF/src/file_ops.c:344:46: error: pointer targets in passing argument 5 of 'RegQueryValueExA' differ in signedness [-Werror=pointer-sign]
[   18s]      if (RegQueryValueEx(hk, "Path", 0, NULL, buf, &len) == ERROR_SUCCESS) {
[   18s]                                               ^
[   18s] In file included from /usr/i686-w64-mingw32/sys-root/mingw/include/windows.h:76:0,
[   18s]                  from /home/abuild/rpmbuild/BUILD/bladeRF-2015.01.rc2.git1422741959.4e78b05/host/libraries/libbladeRF/include/libbladeRF.h:42,
[   18s]                  from /home/abuild/rpmbuild/BUILD/bladeRF-2015.01.rc2.git1422741959.4e78b05/host/libraries/libbladeRF/src/file_ops.c:29:
[   18s] /usr/i686-w64-mingw32/sys-root/mingw/include/winreg.h:183:25: note: expected 'LPBYTE' but argument is of type 'char *'
[   18s]    WINADVAPI LONG WINAPI RegQueryValueExA(HKEY hKey,LPCSTR lpValueName,LPDWORD lpReserved,LPDWORD lpType,LPBYTE lpData,LPDWORD lpcbData);
[   18s]                          ^
[   19s] cc1: all warnings being treated as errors
[   19s] libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/build.make:202: recipe for target 'libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/file_ops.c.obj' failed
[   19s] make[2]: *** [libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/file_ops.c.obj] Error 1

```
worked around that issue by disabling "-Werror"
[buildlog](https://blackhole.pmtu.de/.dump/bladeRF-MinGW-compile-issue)

jynik, 2015-02-14:
Thanks for following up to our recent discussion with this detailed report. 

I think we should be able to wrap the CyAPI usage with a CMake check for MinGW, right? So I'll have to check the manual on that.

For 02, you have a good point, but I suspect we can tackle this with the solution to 01.  With the lack of nice package management in Windows, I was a bit concerned about people building and just copying things around... sticking the licenses in there was a perhaps feeble attempt at making sure folks look at them and include them in their re-distribution of those files.

I'll take a closer look at 03 and why this differs between visual studio and mingw, and whether we can safely cast after an appropriate check/assert.

Any further advice/comments are appreciated.

jynik, 2015-02-16:
I pushed a [dev-issue_374](https://github.com/Nuand/bladeRF/tree/dev-issue_374) branch to address 01 and 02.

For what it's worth, you can disable -Werror via -DTREAT_WARNINGS_AS_ERRORS=OFF (this is a project-specific option).

I'm not seeing the warning (treated as an error) in 03. I've got `i686-w64-mingw32-gcc (GCC) 4.9.1`, from the Ubuntu 14.10 package repo.

I am, however, getting the following warnings, which I still need to look at further:
```
/jon/projects/bladeRF/host/libraries/libbladeRF_test/test_timestamps/src/test_timestamps.c: In function ‘counter_data_is_valid’:
/jon/projects/bladeRF/host/libraries/libbladeRF_test/test_timestamps/src/test_timestamps.c:99:21: warning: unknown conversion type character ‘l’ in format [-Wformat=]
                     (unsigned long long) i, ctr, val);
                     ^
/jon/projects/bladeRF/host/libraries/libbladeRF_test/test_timestamps/src/test_timestamps.c:99:21: warning: format ‘%x’ expects argument of type ‘unsigned int’, but argument 3 has type ‘long long unsigned int’ [-Wformat=]
/jon/projects/bladeRF/host/libraries/libbladeRF_test/test_timestamps/src/test_timestamps.c:99:21: warning: too many arguments for format [-Wformat-extra-args]
/jon/projects/bladeRF/host/libraries/libbladeRF_test/test_timestamps/src/test_tx_onoff.c: In function ‘run’:
/jon/projects/bladeRF/host/libraries/libbladeRF_test/test_timestamps/src/test_tx_onoff.c:124:9: warning: unknown conversion type character ‘l’ in format [-Wformat=]
         printf("Sending burst @ %llu\n", (unsigned long long) meta.timestamp);
         ^
/jon/projects/bladeRF/host/libraries/libbladeRF_test/test_timestamps/src/test_tx_onoff.c:124:9: warning: too many arguments for format [-Wformat-extra-args]
```

jynik, 2015-02-16:
Pushed fixes to the two warnings I saw... still need to look for what you reported in 03.

Mind letting me know what your build system is? I could set up a VM at some point...

jynik, 2015-02-17:
* 01 fixed by d9f290505b476fd22266fec2b12266d7a94e6682
* 02 fixed by 50924f5b4130ac004da3738ce90d676a1d24d9ad
* 03 fixed by 8744904afb5d6f940576cd90b250a10bf08cb0d9
* Other warnings I encountered are addressed in fdd843c36156335d8777120e66f774a99aa10a75

closed 2015-02-17 by jynik