Nuand/bladeRF #375: Build error - missing defines

mambrus, 2015-02-15:
This following build-error has started to manifest.

```
In file included from ${SRC_DIR}/host/utilities/bladeRF-cli/src/common.h:30:0,
                 from ${SRC_DIR}/host/utilities/bladeRF-cli/src/cmd/cmd.h:28,
                 from ${SRC_DIR}/host/utilities/bladeRF-cli/src/cmd/cmd.c:26:
${SRC_DIR}/host/utilities/bladeRF-cli/src/cmd/cmd.c:105:27: error: ‘CLI_CMD_HELPTEXT_calibrate’ undeclared here (not in a function)
         FIELD_INIT(.help, CLI_CMD_HELPTEXT_calibrate),
:
:
(Lots more of the same)
:
```

mambrus, 2015-02-15:
Analyze shows that this is not code (change) related. Checking out an old label previously known to build fine still fails. I.e. issue is build-host related which triggers a weak-spot in the tool-chain.

The problem started after pandoc was installed from source instead as from distro repo. In my case I have no choice than to use the latest pandoc for another project... 

Seems problem is two-fold

* in ``utilities/bladeRF-cli/CMakeLists.txt:31`` (i.e. ``if(PANDOC_EXECUTABLE)``) test fails on Ubuntu 12.04 (?) and the committed backup is used instead, i.e. ``./host/utilities/bladeRF-cli/src/cmd/doc/cmd_help.h.in``. (Side-note: Suffix .in might benefit from change as it implies a autoconf template header and should be stripped/replaced with something less miss-leading). 

* Using committed instead of generated file works fine in most cases (which I'm guessing is the case for Windows builds), but effectively hides the real problem. The code-generator ``generate.bash`` is too format sensitive. If pandoc is installed from source, test will succeed, but the format now is different from what the generator was written for.

Solving the second point alone would solve the issue.

jynik, 2015-02-16:
Just a quick "note-to-self" -- this sounds related to #369

mambrus, 2015-02-16:
#369 has the same root-cause as #375. Analyze is correct but not sure about the solution, or if it solves the same manifestation.

jynik, 2015-02-16:
If see a warning in the Pandoc 1.13 release notes regarding a change potentially breaking existing documents.

Given that our current script is fine with Pandoc 1.12.4.2, and borks with Pandoc 1.13.1, I suspect the 1.13 change may be related to this cropping up.

jynik, 2015-02-17:
Resolved by @mambrus's patch in 89612e7dc361eeb34f65ac224bd84b3f7f2e6f3d

Thank you! :)

closed 2015-02-17 by jynik