Nuand/bladeRF #377: Timeout trying to update from firmware 1.5.3

roger-, 2015-02-19:
I'm getting the following message saying I need to update my firmware:

```
C:\Program Files\bladeRF\x64>bladeRF-cli.exe -i
[INFO @ usb.c:473] FX3 FW v1.5.3 does not support the "device ready" query.
        Ensure flash-autoloading completes before opening a device.
        Upgrade the FX3 firmware to avoid this message in the future.

[WARNING @ bladerf.c:168] Firmware v1.5.3 was detected. libbladeRF v1.0.0 requires firmware v1.6.0 or later. An upgrade via the bootloader is required.

Failed to open device (first available): A firmware update is required
```

Here's my attempt at doing so using the first approach [here](https://github.com/Nuand/bladeRF/wiki/Upgrading-bladeRF-firmware):

```
C:\Program Files\bladeRF\x64>bladeRF-cli -f bladeRF_fw_latest.img -v verbose
[VERBOSE @ libusb.c:339] Using libusb version: <= 1.0.9
[VERBOSE @ libusb.c:347] Found a bladeRF (based upon VID/PID)
[INFO @ usb.c:473] FX3 FW v1.5.3 does not support the "device ready" query.
        Ensure flash-autoloading completes before opening a device.
        Upgrade the FX3 firmware to avoid this message in the future.

[VERBOSE @ usb.c:251] Changing to USB alt setting 0
[VERBOSE @ usb.c:251] Changing to USB alt setting 1
[WARNING @ bladerf.c:168] Firmware v1.5.3 was detected. libbladeRF v1.0.0 requires firmware v1.6.0 or later. An upgrade via the bootloader is required.

Failed to open device (first available): A firmware update is required
```

Okay, so using the second approach:

```
C:\Program Files\bladeRF\x64>bladeRF-cli.exe -i
[INFO @ libusb.c:421] Found FX3 bootloader device on bus=3 addr=15. This may be a bladeRF.
[INFO @ libusb.c:425] Use bladeRF-cli command "recover 3 15 <FX3 firmware>" to boot the bladeRF firmware.
No device(s) attached.
bladeRF> recover 3 15 bladeRF_fw_latest.img

  Success! Use "open" to switch to this device.
  Note that a "load fx3 <firmware>" is required to write the firmware to flash.

bladeRF> open

  Error: Operation timed out
```

I keep getting the final timeout. Any idea what's wrong?

jynik, 2015-02-19:
Did you disable any FPGA autoloading (from flash or software) prior to trying to upgrade the firmware?  If not, you should use the bootloader like you did to load v1.5.3 into the FX3's ram (via `recover`), exit the CLI and then run `bladeRF-cli -L X` to erase the FPGA.  From tere you should be able to use the second approach again smoothly.

The reason for this is described in [this post](https://www.nuand.com/forums/viewtopic.php?f=2&t=3514&start=0), and described in much more detail in [this forum discussion](http://nuand.com/forums/viewtopic.php?f=4&t=3515#p5097).  A similar symptom and brief explanation is also noted [here](https://github.com/Nuand/bladeRF/wiki/Troubleshooting#error-could-not-write-to-si5338--5-file-or-device-io-failure) on the wiki as well.

Basically, in the early FX3 firmware and FPGA images, the UART bridge was using a low baud rate. In FX3 firmware 1.6.0 and FPGA version v0.0.1, this rate was increased to allow control communication to occur quicker. However, this had the downside of requiring that these two versions be updated together, and was a particularly tricky update. (Fortunately we shouldn't be making such a drastic change any time soon...)

Give those links a good read through and let me know if that makes sense.  If you still have trouble, perhaps we should take this discussion to the forums, and you could provide a verbose log (passing `-v verbose` to bladeRF-cli).

There's also a means to tell libbladeRF to assume no FPGA is loaded, and to make no attempt to communicate with it -- you can set the BLADERF_FORCE_NO_FPGA_PRESENT variable in your environment.... so you could give this a shot to erase the FPGA from flash, if "v0.0.0" is present.
`$ BLADERF_FORCE_NO_FPGA_PRESENT=1 bladeRF-cli -L X`

roger-, 2015-02-19:
I'm not sure if someone else may have disabled FPGA autoloading on the unit, but I set the jumper and  followed your instructions and got this (this is on Windows 7):

```
C:\Program Files\bladeRF\x64>bladeRF-cli -i
[INFO @ libusb.c:421] Found FX3 bootloader device on bus=3 addr=27. This may be a bladeRF.
[INFO @ libusb.c:425] Use bladeRF-cli command "recover 3 27 <FX3 firmware>" to boot the bladeRF firmware.
No device(s) attached.
bladeRF> recover 3 27 bladeRF_fw_v1.5.3.img

  Success! Use "open" to switch to this device.
  Note that a "load fx3 <firmware>" is required to write the firmware to flash.

bladeRF> exit

C:\Program Files\bladeRF\x64>bladeRF-cli -L X
[INFO @ usb.c:473] FX3 FW v1.5.3 does not support the "device ready" query.
        Ensure flash-autoloading completes before opening a device.
        Upgrade the FX3 firmware to avoid this message in the future.

[WARNING @ bladerf.c:168] Firmware v1.5.3 was detected. libbladeRF v1.0.0 requires firmware v1.6.0 or later. An upgrade via the bootloader is required.

Failed to open device (first available): A firmware update is required
```

Presumably this means I need to downgrade my host software? If so, which version should I go for?

I also tried the setting `BLADERF_FORCE_NO_FPGA_PRESENT=1`, but I still got the error message about needing to upgrade the firmware.

jynik, 2015-02-19:
Glad you got sorted out!

Yeah I agree --  I think this is worth its own wiki page, linked to from the firmware upgrade page.

jynik, 2015-02-28:
Marking "invalid" just to separate this from items that required source-level changes.