Nuand/bladeRF #386: android usb bindings

jettero, 2015-03-23:
I was using https://github.com/demantz/RFAnalyzer on my phone with a dvb rtl and it's really fun.  The app supports HackRF as well, but not BladeRF.  bpadalino suggested Nuand might be interested in writing android bindings (whatever that might entail); then it might be possible to get bladerf working in the rfanalyzer some day too.

demantz, 2015-03-23:
As I already wrote [here](https://github.com/demantz/RFAnalyzer/issues/22#issuecomment-85177256), I would be interested in writing the Android bindings but I don't own a BladeRF so far. However it shouldn't be hard for a BladeRF owner to implement these bindings by looking at the code of https://github.com/demantz/hackrf_android.

Will the BladeRF operate on a USB 2.0 port (only 500mA)?

jettero, 2015-04-29:
Oh, sorry for replying a month later.  I didn't see this question when you asked it...

Yes, the bladerf works on usb2 ports.  I would think that's enough current to do RX, maybe not TX, but I can't confirm through their documentation.   I suppose I'll have to try it.  I wonder if I could manage to port their bladerf-cli to android so I can talk to it from the shell.

Adding to the todo list.

jynik, 2015-06-09:
@demantz,

Full duplex might be hit or miss on most (USB 2.0) phones without external power.  We might be able to get away with TX -- I'll have to take some voltage measurements.

I have a Galaxy S5, which has a USB 3.0 connector.  (Although I'm a bit disappointed to find that it enumerates at High-speed when attached to a computer as an MTP device...)  I know nothing about Android and USB 3.0, but would be happy to test things out for you.

jettero, 2015-06-10:
I'm not as worried about TX, personally,  I just like the Spec-A interfaces in RFAnalyzer.  I'm going to try to get GNURadio working on my phone at some point — http://gnuradio.org/redmine/projects/gnuradio/wiki/Android — and that would handle TX/RX similarly, etc.  I wonder how the QT/WX interfaces would work in a frame buffer... would they work at all?

I may also work on the bindings myself so I can just use RFAnalyzer … I'm hoping someone else does it before I get to the attempt; because a) time? Where do you buy that? and b) I don't know anything about the various low level thingamajigs, so it's going to take me a while to make even trivial progress if I do find the time.

demantz, 2015-06-12:
Oh, I missed that GNU Radio is now able to build for Android. Very interesting - thanks for the link!

Regarding the Android bindings for the BladeRF:
Unfortunately I still do not own a BladeRF yet in order to make some first attemps. But I would offer my help if someone tries to implement the bindings for Android! And maybe at some point in the future I will purchase a BladeRF..

The RF Analyzer app is RX only. So it might actually work. Even if it won't get enough power from some Android devices there is still the possibility to use a powered USB hub.

jettero, 2015-06-12:
Yeah, my thoughts exactly.  TX isn't needed for RF Analyzer.  But I think the notion with making a library is to possibly open development generically on Android rather than just in RFA.

jynik, 2015-06-15:
@demantz: I'm not terribly familiar with Android development, so pardon the ignorance in my questions and comments on the matter.

If possible, I'd prefer it if we were able to compile libbladeRF for Android rather than reimplement the entire lib using the Java interfaces.  I think this would have the advantage of users benefiting from fixes and additions. From very basic searches, it sounds like this is possible via the NDK.  Does this seem feasible, or are there a lot of "ifs and buts" about it that I probably didn't dig into yet?

With regard to the analyzer, we should have some fun scheduled-retune functionality in master soon, allowing you do a nice wideband sweep!

demantz, 2015-06-15:
@jynik: Yes this is a good point! In general, compiling libbladeRF on Android using the NDK should be possible.

The main reason I chose to reimplement the hackrf library in Java on Android was because I wasn't familiar with the NDK (actually I'm still not). Plus I did learn a lot about the HackRF and its internals by doing so.
The advantage of having bindings in pure Java is that it would be very easy to use for people that don't want to work with the NDK. But I agree with you that reimplementing libbladeRF would have many disadvantages. It is more complex than libhackrf.

The rtl-sdr driver (and libusb) have been compiled for Android using the NDK: https://github.com/martinmarinov/rtl_tcp_andro-
I am using this Android version of the rtl-sdr driver in RF Analyzer.

jettero, 2015-06-16:
The NDK is definitely an option.  I ran through the GNURadio for Android build process this weekend — and determined it does nothing useful for me … but I also learned that it uses the NDK and the Android SDK build tools extensively.  It could possibly be a model for how to port the lib or at least what commands need to be run to compile it (maybe without modification).

jynik, 2015-06-17:
@jettero Nice job. Perhaps you can share your experiences in more detail on the forums? :)

If you want to give compiling libbladeRF a shot with the NDK, I would suggest the following build flags to only build a dummy "does nothing" backend, and not compile a USB backend.  This will at least let us know which non-USB-related bits will need to be fixed/adjusted, prior to tackling anything USB-related:

```
cmake -DENABLE_BACKEND_USB=OFF -DENABLE_BACKEND_DUMMY=ON ../
```

jettero, 2015-06-18:
I'm not totally sure what you mean.  I do a lotta code, but it's mostly SRE or Network Ops type code… I believe you're talking over my head.  On the otherhand, I should have a lot of time weekend after next and I find with these projects it's really only a matter of how much time you can throw at them.

As far as discussion goes, my experience is the following:  0. read wiki, 1. do thing it says with minor adjustments to path variables, 2. goto 0

I couldn't get the Apache Thrift stuff to build.  Spent about a hour on it and gave up — I can't figure out what it actually is, so I didn't know if I'd actually want it.   As far as I can tell, the GNURadio for Android is mainly UHD; though, it had me build RTL-SDR as a prereq.  I don't use my BladeRF enough to remember (been mainly doing USRP things lately), but I seem to recall that BladeRF uses RTL-SDR … so perhaps I should spend more time on it to see if BladeRF can be made to work out of the box.  Having said that, I seem to recall (I got almost no sleep this week, so I can't really remember) that there was some show-stopper in the RTL-SDR build or somewhere in the directions that led me to believe it wasn't going to work with the BladeRF.  Looking at it on the wiki now though, it seems like it should work.

Perhaps I'll go through their Build-an-App examples and see if I can get a little Spec-A activity to come up.  If I can do it, I'd obviously post that to Github here.  How would that affect this ticket?  I seem to have a working android libusb and a built RTL-SDR, and some other things.  Hrm.

jynik, 2015-06-18:
@jettero,

I think the CMake options (just to be pedantic, they're not path variables) above will make a little more sense after you have a chance to refresh yourself on the process for building libbladeRF from source.  (I'd do it on a host machine first just to get your bearings.)

I believe GNU Radio is using Thrift in their reborn support for [Control Ports](https://gnuradio.org/redmine/projects/gnuradio/wiki/ControlPort).

For bladeRF GNU Radio support, you'll need [gr-osmosdr](http://sdr.osmocom.org/trac/wiki/GrOsmoSDR).  This is a library and generic block that supports a variety of SDRs, including the bladeRF and RTL-SDR.

I'm not sure what build dep they had, but you should not need anything RTL-SDR related from the bladeRF side of things.

The first step we need it to verify whether libbladeRF builds on with the NDK and if there are things we need to fix.  Fortunately, this can be done in parallel to your GNU Radio build efforts, so perhaps you can ping-pong when one becomes too much of a pain. ;)

jettero, 2015-06-19:
right.  osmo.  I don't think I got osmosdr to build.  can't recall why either.   Maybe I'll look at that today during my lunch break.  Maybe try to build the libbladerf too.

jettero, 2015-06-19:
Oh, and the oth

jettero, 2015-06-19:
Ahh, interestingly, the trouble I was having building gr-osmosdr was with the UHD include files, which I totally don't need for the BladeRF.  Also, it was UHD that needed RTL-SDR.  I changed -DENABLE_UHD=True to False and -DENABLE_BLADERF=True and now it stopped complaining about UHD and instead complains about libbladeRF.  Tada!! \o/

Next I'm off to try to build libbladeRF … possibly with the cmake flags you provided if it doesn't build.

jettero, 2015-06-22:
I didn't spend a ton of time on it, but I can't get libbladeRF to compile due to missing pthread libs.  I think they're there, the build doesn't agree.  I can't tell yet if it's a path problem or if the sysroot lib format is messing up the compiler or something.

jynik, 2015-06-22:
Mind sharing some logs of the failure?

jettero, 2015-06-22:
Oh, interesting.  It's not what I thought at all.  All the other pthread functions exist.  In fact, pthread_cancel is only called very occasionally and it's the only one failing to exist — under -Werror it's a dealbreaker, though it clearly wouldn't work without it.

http://stackoverflow.com/a/26824177/388466

The (presumably) android dev said pthread_cancel() is always harmful.  I have no opinion on that, but at least I know what I'm up against.  For build-testing purposes, I'll just #define pthread_cancel(...) /* pthread_cancel() woot! */

jettero, 2015-06-22:
Oh, sure, logs.  They're not very interesting... it's just building building building and then this:

    [ 29%] Building C object host/libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/sync_worker.c.o
    cd /home/jettero/code/android/grandroid/bladeRF/build/host/libraries/libbladeRF && /usr/local/android/standalone-v10-ndk/bin/arm-linux-androideabi-gcc  -DANDROID -DENABLE_USB_DEV_RESET_ON_OPEN=1 -DLOGGING_ENABLED -DLOG_INCLUDE_FILE_INFO -Dlibbladerf_shared_EXPORTS -Wno-psabi --sysroot=/usr/local/android/standalone-v10-ndk/sysroot -fpic -funwind-tables -finline-limit=64 -fsigned-char -no-canonical-prefixes -march=armv7-a -mfloat-abi=softfp -mfpu=vfpv3-d16 -fdata-sections -ffunction-sections -Wa,--noexecstack -Wno-psabi --sysroot=/usr/local/android/standalone-v10-ndk/sysroot -fpic -funwind-tables -finline-limit=64 -fsigned-char -no-canonical-prefixes -march=armv7-a -mfloat-abi=softfp -mfpu=vfpv3-d16 -fdata-sections -ffunction-sections -Wa,--noexecstack  -fvisibility=hidden -mthumb -fomit-frame-pointer -fno-strict-aliasing -mthumb -fomit-frame-pointer -fno-strict-aliasing -O3 -DNDEBUG -fPIC -isystem /usr/local/android/standalone-v10-ndk/sysroot/usr/include -isystem /usr/local/android/standalone-v10-ndk/include/c++/4.8 -isystem /usr/local/android/standalone-v10-ndk/include/c++/4.8/arm-linux-androideabi/armv7-a -I/home/jettero/code/android/grandroid/bladeRF/host/libraries/libbladeRF/include -I/home/jettero/code/android/grandroid/bladeRF/host/libraries/libbladeRF/src -I/home/jettero/code/android/grandroid/bladeRF/host/common/include -I/home/jettero/code/android/grandroid/bladeRF/build/host/common/include -I/home/jettero/code/android/grandroid/bladeRF/host/../firmware_common -I/home/jettero/code/android/grandroid/bladeRF/host/../fpga_common/include -I/home/jettero/code/android/grandroid/bladeRF/build/host/libraries/libbladeRF/src    -Wall -Wextra -Wno-unused-parameter -Werror -DSHORT_FILE_="sync_worker.c" -o CMakeFiles/libbladerf_shared.dir/src/sync_worker.c.o   -c /home/jettero/code/android/grandroid/bladeRF/host/libraries/libbladeRF/src/sync_worker.c
    /home/jettero/code/android/grandroid/bladeRF/host/libraries/libbladeRF/src/sync_worker.c: In function 'sync_worker_deinit':
    /home/jettero/code/android/grandroid/bladeRF/host/libraries/libbladeRF/src/sync_worker.c:256:9: error: implicit declaration of function 'pthread_cancel' [-Werror=implicit-function-declaration]
             pthread_cancel(w->thread);
             ^
    cc1: all warnings being treated as errors
    make[3]: *** [host/libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/sync_worker.c.o] Error 1
    make[3]: Leaving directory `/home/jettero/code/android/grandroid/bladeRF/build'
    make[2]: *** [host/libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/all] Error 2
    make[2]: Leaving directory `/home/jettero/code/android/grandroid/bladeRF/build'
    make[1]: *** [host/utilities/bladeRF-cli/CMakeFiles/bladeRF-cli.dir/rule] Error 2
    make[1]: Leaving directory `/home/jettero/code/android/grandroid/bladeRF/build'
    make: *** [bladeRF-cli] Error 2

jynik, 2015-06-22:
pthread_cancel() is only used as a last-ditch effort in shutting down a worker thread when it appears to have gone off into lala land.  ([Link to this in sync_worker.c](https://github.com/Nuand/bladeRF/blob/7b0cc24f950f3c9a2e8d39080e4dffc75a511f40/host/libraries/libbladeRF/src/sync_worker.c#L256))

You may see a similar approach in some of the libbladeRF_test_* programs as well.

I'd say that ifdef'ing this out is fine here.  I imagine we can move forward with this is adding a BLADERF_OS_ANDROID option to our top-level CMake, and then leverage that.

In touching anything thread-related, I've been leaning toward tucking it behind a implementation an-implementation agnostic set of macros (or static inline fns) in [thread.h](https://github.com/Nuand/bladeRF/blob/7b0cc24f950f3c9a2e8d39080e4dffc75a511f40/host/common/include/thread.h), just to keep all the cruft in one spot.  (Eventually I'd like to drop pthreads as a Windows dep in favor of native Windows threads, just because we use so little of pthreads.)

jettero, 2015-06-22:
indeed, maybe a simple fprintf(stderr, "dood!! is is crashing\n"); exit(1);

mambrus, 2015-06-24:
I'm pretty certain pthread_cancel exists on Android. Their libc is quite
mature wrt POSIX 1003.1c and this doesn't look like a linking error. What
CMake TOOL_CHAIN file was used and is sysroot correct?

Regards //Michael

jettero, 2015-06-24:
From a comment in standalone-v10-ndk/include/c++/4.8/arm-linux-androideabi/armv7-a/thumb/bits/gthr-posix.h:

    The bionic (Android) C
    library does not provide pthread_cancel, so we do use pthread_create
    there (and interceptor libraries lose).

It's not a matter of maturity.  It was a conscious decision made by the Bionic team, or so I currently believe. No, it's not a linking error.  It's a warning that the function isn't defined and -Werror is active, so it never gets past that point.

I never did try to provide it as a comment-macro or replace it with fprintf(stderr,...); exit(1) or whatever.  I'm fairly certain that will work.  I just didn't get to it.

mambrus, 2015-06-24:
I created a new repo to experiment with stuff such as build-systems and
target libraries. Have a look at the following:

https://github.com/mambrus/mixedmisc/tree/master/posixtest_android

But, it proves me wrong. I couldn't believe it so checked four-ways:

   - Trial and error using the above project (please read gitlog as well)

   - Sometimes NDK hide stuff they dont think should be available so I
   searched for the symbol in Bionic:
   http://androidxref.com/5.1.0_r1/search?q=pthread_cancel&defs=&refs=&path=&hist=&project=bionic
   - Checked one of my projects where I knew this was used, and to my
   surprise: Where pthread_cancel was used, someone rewrote the code to work
   around the problem (grr).
   - Googled it:
   http://stackoverflow.com/questions/12111640/android-ndk-and-posix-functions

I certainly don't agree with the reasons for this limitation, not usingt
this as it's always "bad code" is something only an Android dev could say.
But there you have it in black and white ;-(

I'll dig deeper asap.
Meanwhile, even though I hate to see *any* changes in code due to standard
deviance, I'm afraid that to get around this quickly, one would have to
work around it (i.e. rewrite code).

Regards //Michael

P.S. Jon et al: Please look at the build-mechanism and let me know what you
think. There's another way using pre-made TOOL_CHAIN files, but I thought
this was sexier as I can reuse the Google compiler for other targets (as
long as the arch and abi are compatoble, which many are). D.S.

~-~ ~-~ ~-~ ~-~
Best Regards
Michael Ambrus
PGP key <http://ambrus.se/michael_ambrus.pub>

*This email and any files transmitted with it are confidential and intended
solely for the use of the individual or entity to whom they are addressed.
If you have received this email in error please notify the system manager.
This message contains confidential information and is intended only for the
individual named. If you are not the named addressee you should not
disseminate, distribute or copy this e-mail. Please notify the sender
immediately by e-mail if you have received this e-mail by mistake and
delete this e-mail from your system. If you are not the intended recipient
you are notified that disclosing, copying, distributing or taking any
action in reliance on the contents of this information is strictly
prohibited. *

jettero, 2015-06-24:
I'm pretty sure platform tuning is why C macros exist in the first place.  I don't see it as such a horrific standards problem as much as it simply requires a build flag or build-time patch or something.

jettero, 2015-06-24:
I injected a -Dpthread_cancel=exit and I compile all the way up to "Linking C shared library ../../output/libbladeRF.so" but it fails with:  "function bladerf_set_usb_reset_on_open: error: undefined reference to 'bladerf_usb_reset_device_on_open'"

I find this odd since I used jynik's "-DENABLE_BACKEND_USB=OFF -DENABLE_BACKEND_DUMMY=ON"  — strictly speaking, I found ON and OFF to not work and actually used False and True.

I'm encouraged by how far I got so far though.  It almost gets to the linking phase.  I think I'll disable the dummy stuff and try to link against libusb (which I have already built and should be able to link against).

jettero, 2015-06-24:
Oh, apparently I could have just used "-DENABLE_USB_DEV_RESET_ON_OPEN" ... ahh well.  Working on the libusb stuff instead.

jynik, 2015-06-24:
Ah, that smells like a mix of options causes a build failure. I'll regard that as a bug. Could you open a separate tracker item and include your CMakeCache.txt from a build that reproduces the failure?

jettero, 2015-06-24:
jynik: I will do it.  Though, I have to go to a bunch of meetings like 4 minutes ago, so I may forget.

I successfully built and then installed libbladerf to my /opt/ dir.  No idea if it works, but it's built and linked against libusb using the android compiler and instructions from GNURadio Android edition.

I posted my build script (literally everything you need, even the pthread_cancel fix is in this script and works on a vanilla download):

    https://goo.gl/yfUatK

jynik, 2015-06-30:
@jettero ...and see if it runs... Don't leave me hanging man! :)

jettero, 2015-06-30:
Haha, forgot about that.  It's non-trivial to try … I somehow have to get all those libs and supporting files and the binary on my phone in such a way that the binary can find the shared libs, etc.  I imagined I'd have to build an android manifest and bundle an APK … then moved on to more glorious projects that don't work.

Perhaps I'll find time today.

jynik, 2015-06-30:
@jettero,

How do you feel about me closing this issue, and moving our notes/discussion on the process over to the wiki and/or a forum thread.

From an issue tracking perspective, I think it will be easier for me to manage TO DO's on my end if we have very fine-grained and specific build issues, such as an item for #ifdef'ing out that pthread_cancel() for android builds.

jettero, 2015-06-30:
Sure, go for it.

mambrus, 2015-06-30:
If you can get an alternative console up, i.e. alternative to adb shell which is USB based as you probably can't tunnel, you don't need an APK. Just copy binary and shared libraries to somewhere you have rights to push to, and right to execute from and set LD_LIBRARY_PATH. Android RTL should honor it. From there you should be able to run at least bladerf-cli. Fully integrated with the system you can't unless you have a rooted phone. But that is of no real concern as you can bundle shared libraries with APK when time comes instead.

The location /data/local/tmp is what I usually use.

If you have a NEXUS 5 there is a apparently a way to get a serial console via the earplug connector. Or even better: use one of those ready-made generic ARM-boards that has Android supported. This for example could work: https://www.youtube.com/watch?v=QdwaaMZBhjo. Or ODROID-XU3, that HW comes as close as you can to the performance of a modern smart-phone.

jynik, 2015-06-30:
I'll close this once I have some time to move the information here to the wiki so that we can maintain a living document of these efforts.  (Feel free to beat me to it!)

From there, why don't we register fine-grained issues and link to them.

jettero, 2015-07-02:
Did this get moved? URL?

jynik, 2015-07-05:
Didn't move anything yet. Will make note of it here once I get around to it!

jettero, 2015-07-06:
OK, no rush. I just didn't want to be missing out on its new location.

jynik, 2015-08-04:
Created a [Building libbladeRF for Android](https://github.com/Nuand/bladeRF/wiki/Building-libbladeRF-for-Android) page on the wiki.

I'll let you guys take the reins regarding adding notes there.

closed 2015-08-04 by jynik