Nuand/bladeRF #390: Warning after running rx start when loopback=firmware

jenselo, 2015-05-25:
When setting loopback to firmware and running rx start, the message "[WARNING @ usb.c:1392] FX3 reported error=0x44 when disabling RF RX" shows up and rx shows the last error as "Operation timed out"

bladeRF-cli reports version as:
  bladeRF-cli version:        1.1.2-git-73cdb21
  libbladeRF version:         1.2.1-git-73cdb21

  Firmware version:           1.8.0
  FPGA version:               0.1.2

Steps to reproduce:
bladeRF> set loopback firmware
bladeRF> rx config file=samples.csv format=csv n=1000
bladeRF> rx start
bladeRF> [WARNING @ usb.c:1392] FX3 reported error=0x44 when disabling RF RX

jynik, 2015-06-09:
The firmware loopback will require both the RX and TX modules to be running in order for samples flow.  It simply takes a DMA buffer from an OUT EP and ships it back to the IN EP.  Since there's only a couple buffers, RX and TX pretty much need to be in lockstep for this to work.

So if you start up RX and don't start up TX in time, then I could see why it would time out.

However, the error=0x44 is something I'll need to look into.  If I remember correctly, this may be an error reported because we're attempting to disable/enable something that's already disabled/enabled. (In which case it's harmless, but an indicator that we should review the logical flow of things...)

I'll be getting back to this shortly. If you have time to dive into it -- checking out what 0x44 is in the CyAPI codes would be good to know.

jenselo, 2015-08-02:
I just noticed that a comment in [usb.c](https://github.com/Nuand/bladeRF/blob/c917d33dbc5ab762ecc03151e0a76436a2267663/host/libraries/libbladeRF/src/backend/usb/usb.c#L851-L859) explains it: 

0x44 corresponds to CY_U3P_ERROR_ALREADY_STARTED

closed 2018-09-13 by rtucker