Nuand/bladeRF #406: Persistant timeouts after using API

roger-, 2015-07-17:
I was playing with the API and called `bladerf_set_sample_rate()` with parameter 100e6, which seems to have messed up my device because I'm now getting constant timeouts when I try to read samples with `bladerf_sync_rx()` or do various other things. Everything worked fine before.

I've tried switching USB ports, calling `bladerf_device_reset()` and reflashing the FPGA, but nothing helps. Calls like `bladerf_get_device_list()` seem to work though.

I also get a timeout with the CLI when trying to calibrate:

```
bladeRF> calibrate dc rxtx
...
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x19 to addr 0xa0
[VERBOSE @ lms.c:1774] VTUNE_BUSY_WAIT(25)
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x1a from addr 0x43
[VERBOSE @ lms.c:1905] VTUNE low @ VCOCAP=32
[VERBOSE @ lms.c:2067] VTUNE LOW:   32
[VERBOSE @ lms.c:2068] VTUNE NORM:  26
[VERBOSE @ lms.c:2070] VTUNE Est:   25 (-1)
[VERBOSE @ lms.c:2071] VTUNE HIGH:  21
[VERBOSE @ lms.c:1785] Writing VCOCAP=26
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x19 to addr 0x9a
[VERBOSE @ lms.c:1774] VTUNE_BUSY_WAIT(25)
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x1a from addr 0x03
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x09 from addr 0x45
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x09 to addr 0x40
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x08 from addr 0x00
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x46 from addr 0x00
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x44 from addr 0x03
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x44 to addr 0x0b
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x46 from addr 0x00
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x08 from addr 0x00
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x46 to addr 0x00
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x08 to addr 0x00

  Error: Operation timed out
```

My version info:

```
bladeRF> version

  bladeRF-cli version:        1.2.0
  libbladeRF version:         1.4.0

  Firmware version:           1.8.0
  FPGA version:               0.3.3
```

(On Windows 7, x64.)

jynik, 2015-07-17:
Hi there,

It sounds like we might be still in the debugging phase here, rather than have a specific bug for this issue tracker.  As such, I'm going close this issue and kindly request that we take this discussion over to the 
[Troubleshooting forum](https://www.nuand.com/forums/viewforum.php?f=4&sid=e50b314a40cc02062320740e91e10f51, where I and some others can lend a hand.

If we identify some specific defects the codebase, we can open appropriate bug reports here.

Although 100e6 is way too high to attempt sampling from the LMS6002D, I don't think that should have damaged the device.  While I don't consider it a solution, have you tried rebooting your machine and just 

When you open a forum thread, would you mind including output for the following, from a fresh power-on of the board?

```
> bladeRF-cli -i -v verbose
bladeRF> print samplerate
bladeRF> print frequency
bladeRF> set samplerate 2M
bladeRF> rx config file=test.bin n=2M
bladeRF> rx start; rx; rx wait 5s;
```

There's a bug in this tracker regarding a calibration timeout  which I'll be hoping to look at soon -- so let's avoid using that in your diagnostics until we get your board back up and running.

jynik, 2015-07-17:
Although not a solution, does your board begin responding after rebooting your machine?

If we can reproduce something that puts a board into a bad state...that's certainly something we'll want to put up here on the tracker, with the exact steps to reproduce.

closed 2015-07-17 by jynik

roger-, 2015-07-17:
Just rebooted and no, still getting the timeout with my code as well as with `rx start; rx; rx wait 5s;` in your last message.