Nuand/bladeRF #407: Setting Loopback resets/changes Rx module frequency

polygon, 2015-07-21:
I've found that setting the Loopback mode to None changes the Rx frequency that is set. An example in bladeRF-cli. The scenario is that an external frequency generator that is set to produce a sine at 2.631 GHz is attached to the Rx-Port of a BladeRF. The FPGA image has already been loaded into the BladeRF but no other options were set, so it is essentially in power-up state + FPGA loaded.

A sample bladeRF-cli session demonstrating the problem:

    $ bladeRF-cli -i
    bladeRF> set frequency 2630000000
      Set RX frequency: 2629999999Hz
      Set TX frequency: 2629999999Hz
    bladeRF> set bandwidth 5000000
      Set RX bandwidth - req:  5000000Hz actual:  5000000Hz
      Set TX bandwidth - req:  5000000Hz actual:  5000000Hz
    bladeRF> set samplerate 5000000
      Setting RX sample rate - req:   5000000 0/1Hz, actual:   5000000 0/1Hz
      Setting TX sample rate - req:   5000000 0/1Hz, actual:   5000000 0/1Hz
    bladeRF> set rxvga1 5
    bladeRF> set rxvga2 5
    bladeRF> rx config file=rxsine.bin format=bin n=10K
    bladeRF> rx start

At this point, taking the received file, loading it in python and running scipy.signal.welch over it produces something like this:

[image]

 This is exactly what is expected. A peak at roughly 1 MHz off-center. Continuing the bladeRF-cli session as follows:

    bladeRF> set loopback none
    bladeRF> rx start

Doing the same processing as before reveals this:

[image]

The peak is gone. I'm not sure what exactly happens but the problem can be fixed by setting the carrier frequency of the BladeRF again.

    bladeRF> set frequency 2630000000
      Set RX frequency: 2629999999Hz
      Set TX frequency: 2629999999Hz
    bladeRF> rx start

Now, processing the data again produces this spectrum:

[image]

Discovered that problem when running a program I wrote about a year ago. It sets the radio loopback mode as part of the configuration process. I know that one year ago this effect didn't happen and I'm wondering if it's intentional.

jynik, 2015-07-21:
Hi @polygon, could you confirm that this is not the same issue as #405, which was fixed in 093aed7288cb0c566cf72af9c3cecca555688de7?  It sounds very similar.  

The cause of #405 was that a regression caused us to always select LNA1 when coming out of loopback mode.  This implies that frequencies under 1.5 GHz would work fine, but those above would exhibit symptoms you've described.

By the way, thank you for putting together an excellent writeup of what you're seeing.  You've set a precedent for what I now want to see in bug reports ;).

polygon, 2015-07-21:
@jynik Apparently that was the problem. Problem does not appear with current master.

closed 2015-07-21 by polygon