Nuand/bladeRF #412: Calibration often fails

polygon, 2015-07-22:
I'm wondering what the current state of DC and IQ-Imbalance calibration is. I'm currently using `cal lms` from bladeRF-cli to calibrate the DC offsets of the several components which usually works flawlessly.

However, using the `cal dc rxtx` command afterwards more often than not fails like so:

    bladeRF> cal dc rxtx
      RX DC I Setting = -2048, error ~= -10
      RX DC Q Setting = -853, error ~= 0
      Error: Obtained out-of-range TX I DC cal value (-3066.523926).
      Error: An unexpected error occurred

Or so:

    bladeRF> cal dc rxtx
      RX DC I Setting = -2048, error ~= -10
      RX DC Q Setting = -853, error ~= 0
      Error: Invalid m01 (0.000006) or m23 (-0.000005) encountered.
      Error: An unexpected error occurred

The BladeRF did not have anything attached to the Tx and Rx ports, was freshly started and just got the FPGA image loaded.

Frequency was set to 2.63 GHz, Txvga1 = -10, Txvga2 = 18, Rxvga1 = Rxvga2 = 5, 5 MHz Samplerate and Bandwidth.

Using just `cal lms` I see about 41dB between the carrier and a single tone at maximum amplitude.

bpadalino, 2015-07-22:
Can you run this again with -v verbose option and attach the log?

polygon, 2015-07-23:
This is the verbose log of just the `cal dc rxtx` command

```
[VERBOSE @ device_identifier.c:134] Requested serial number: d789da370687895f0eefafc8a574d669
[VERBOSE @ libusb.c:515] Using libusb version: 1.0.19.10903
[VERBOSE @ libusb.c:406] Found a bladeRF (idx=4)
[VERBOSE @ libusb.c:458] USB port reset succeeded for bladeRF d789da370687895f0eefafc8a574d669
[VERBOSE @ capabilities.c:41] Capability mask before FPGA load: 0x0000000300000000
[VERBOSE @ usb.c:112] Changing to USB alt setting 0
[VERBOSE @ usb.c:112] Changing to USB alt setting 1
[VERBOSE @ usb.c:169] Read FPGA version: 0.3.3
[VERBOSE @ capabilities.c:72] Capability mask after FPGA load: 0x000000030000003f
[VERBOSE @ usb.c:180] Using current packet handler formats
[VERBOSE @ usb.c:112] Changing to USB alt setting 2
[VERBOSE @ usb.c:112] Changing to USB alt setting 1
[VERBOSE @ usb.c:112] Changing to USB alt setting 2
[VERBOSE @ usb.c:112] Changing to USB alt setting 1
[VERBOSE @ nios_access.c:664] Clearing RX retune queue.
[VERBOSE @ nios_access.c:687] RX operation duration: 0
[VERBOSE @ nios_access.c:664] Clearing TX retune queue.
[VERBOSE @ nios_access.c:687] TX operation duration: 0
[VERBOSE @ nios_access.c:298] nios_config_read: Read 0x0000002f
[DEBUG @ tuning.c:82] Default tuning mode: FPGA
[VERBOSE @ nios_access.c:298] nios_config_read: Read 0x0000002f
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x54 from addr 0x26
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x20 from addr 0x44
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x21 from addr 0x7d
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x22 from addr 0x55
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x23 from addr 0x55
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x25 from addr 0xb2
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x29 from addr 0xb0
[VERBOSE @ nios_access.c:298] nios_config_read: Read 0x0000002f
[VERBOSE @ si5338.c:280] Reading MS1
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x25 from addr 0x01
[VERBOSE @ si5338.c:289] Read enable register: 0x01
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x40 from addr 0xb8
[VERBOSE @ si5338.c:298] Read regs[0]: 0xb8
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x41 from addr 0x7c
[VERBOSE @ si5338.c:298] Read regs[1]: 0x7c
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x42 from addr 0x20
[VERBOSE @ si5338.c:298] Read regs[2]: 0x20
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x43 from addr 0x00
[VERBOSE @ si5338.c:298] Read regs[3]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x44 from addr 0x00
[VERBOSE @ si5338.c:298] Read regs[4]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x45 from addr 0x00
[VERBOSE @ si5338.c:298] Read regs[5]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x46 from addr 0x19
[VERBOSE @ si5338.c:298] Read regs[6]: 0x19
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x47 from addr 0x00
[VERBOSE @ si5338.c:298] Read regs[7]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x48 from addr 0x00
[VERBOSE @ si5338.c:298] Read regs[8]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x49 from addr 0x00
[VERBOSE @ si5338.c:298] Read regs[9]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x20 from addr 0xc0
[VERBOSE @ si5338.c:308] Read r register: 0xc0
[VERBOSE @ si5338.c:147] Unpacked P1: 0x00007cb8 (31928) P2: 0x00000008 (8) P3: 0x00000019 (25)
[VERBOSE @ si5338.c:165] Unpacked a + b/c: 253 + 11/25
[VERBOSE @ si5338.c:166] Unpacked r: 1
[VERBOSE @ si5338.c:332] Calculated samplerate: 5000000 + 0/1
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x54 from addr 0x26
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x54 to addr 0x26
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x55 from addr 0x15
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x54 from addr 0x26
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x54 to addr 0x3e
[VERBOSE @ si5338.c:477] Setting integer sample rate: 3000000
[VERBOSE @ si5338.c:364] Found r value of: 1
[VERBOSE @ si5338.c:374] MSx a + b/c: 422 + 2/5
[VERBOSE @ si5338.c:403] MSx a + b/c: 422 + 2/5
[... 1122 lines cut ...]
[VERBOSE @ lms.c:2068] VTUNE NORM:  49
[VERBOSE @ lms.c:2070] VTUNE Est:   48 (-1)
[VERBOSE @ lms.c:2071] VTUNE HIGH:  44
[VERBOSE @ lms.c:1785] Writing VCOCAP=49
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x19 to addr 0xb1
[VERBOSE @ lms.c:1774] VTUNE_BUSY_WAIT(25)
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x1a from addr 0x03
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x09 from addr 0x45
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x09 to addr 0x40
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x08 from addr 0x00
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x46 from addr 0x00
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x44 from addr 0x03
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x44 to addr 0x13
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x46 from addr 0x00
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x08 from addr 0x00
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x46 to addr 0x00
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x08 to addr 0x00

  Error (cal_dc_rxtx.txt:0): An unexpected error occurred

[VERBOSE @ nios_access.c:664] Clearing RX retune queue.
[VERBOSE @ nios_access.c:687] RX operation duration: 0
[VERBOSE @ nios_access.c:664] Clearing TX retune queue.
[VERBOSE @ nios_access.c:687] TX operation duration: 0
```

jynik, 2015-07-24:
The following messages occur when the gain+frequency combination is such that the LMS6002D's DC correction registers are maxed out before it can pull the signal to 0. 
(Aside from that, the script line being 0 is weird to me... need to look at that.)

```
  Error (cal_dc_rxtx.txt:0): Invalid m01 (0.000015) or m23 (-0.000004) encountered.
```

@polygon, could you confirm that my above assertion is indeed true?  One way to do this is to run `osmocom_fft -S --dc-cal-mode=1 -s <samplerate>`.  If you're able to move I and Q up to or across the zero crossing, my above statement is wrong and we have more debugging to do.

If my assertion is true, we should have the CLI print something more articulate -- complaining that it failed to find the intersection of two linear interpolations is certainly not helpful to the end user.

I don't like the ``` Error: Obtained out-of-range TX I DC cal value (-3066.523926).``` from the first log...that's something I'll want to look into....

polygon, 2015-07-27:
@jynik You are correct, the I slider does not have a "zero", it seems to decrease the closer it goes to becoming `1` but never increases again. It also seems still quite far away from reaching any kind of optimum (Moving the Q slider does not to much at all). Main question is "why", though.

ttlewi, 2015-08-05:
I am seeing a similar issue following the same steps @polygon has, only <[DEBUG @ sync_worker.c:95] RX overrun @ buffer 1> is printed a couple of hundred times during the tx dc calibration.

ttlewi, 2015-08-05:
Update: I switched drivers from the cypress driver to libusb and magically everything works!
```
bladeRF> bladeRF-cli.exe -i
bladeRF> info

  Serial #:                 881b7a2b6eb836e51effafad641813c1
  VCTCXO DAC calibration:   0x8da1
  FPGA size:                115 KLE
  FPGA loaded:              yes
  USB bus:                  2
  USB address:              31
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

bladeRF> version

  bladeRF-cli version:        1.2.0
  libbladeRF version:         1.4.0

  Firmware version:           1.8.0
  FPGA version:               0.3.3

bladeRF> cal lms

  Calibrating LMS LPF tuning module...
    LPF tuning module: 23

  Calibrating LMS TX LPF modules...
    TX LPF I filter: 37
    TX LPF Q filter: 27

  Calibrating LMS RX LPF modules...
    RX LPF I filter: 1
    RX LPF Q filter: 5

  Calibrating LMS RXVGA2 modules...
    RX VGA2 DC reference module: 25
    RX VGA2 stage 1, I channel: 23
    RX VGA2 stage 1, Q channel: 25
    RX VGA2 stage 2, I channel: 35
    RX VGA2 stage 2, Q channel: 35

bladeRF> cal dc rx

  RX DC I Setting = 242, error ~= 0
  RX DC Q Setting = 160, error ~= -1

bladeRF> cal dc tx

  TX DC I Setting = -79, error ~= 6.728403
  TX DC Q Setting = 476, error ~= 3.426508
```

chackn, 2015-08-05:
I experienced the same trouble with cypress USB driver.

polygon, 2015-08-07:
I was using libusb all the time, your issue might be related, but probably different.

jynik, 2015-08-27:
@polygon Regarding - "why" -- this is just the limitation of the LMS6002D - the compensation DACs have a limited range.

As far as handling this, I think in the CLI we'll have to just warn that this was the case rather than bail out.  In this situation, the user might want to consider using different gains (bearing in mind that there may not be a silver bullet set of gains for every frequency in the tunable range).

jynik, 2015-12-29:
The problems reported here have been taken into account in the re-design of the DC cal routines, thus I am marking this closed as of 1154ef7bd99b909a2eed9893afc999b553c94fe9.

If anyone should find issues in the latest and greatest, let's open a new issue to track it with respect to the new DC cal routine implementations.

closed 2015-12-29 by jynik