Nuand/bladeRF #414: 10 MHz reference clock support

polygon, 2015-07-29:
About a year ago there were several threads in the forums about supporting a 10 MHz reference clock input since it is often used to synchronize frequencies of laboratory equipment and good 10 MHz sources are widely available and also cheaper. For example, I am now using a 10 MHz synced frequency generator to produce the 38.4 MHz required by the BladeRF which is quite the overkill especially since I could need that device elsewhere, too.

Maybe the devs could give a short status on whether this is possible at all and we could use this ticket to track progress.

I'm also interested in not too expensive solutions to turn a 10 MHz reference into a 38.4 MHz reference. There is probably some clock synthesizer evaluation boards. Does anyone have experience here on solutions that work properly?

bpadalino, 2015-08-04:
We have not written the control loop to do the counting of the clocks and comparisons along with setting the VCTCXO.  A register interface to the NIOS along with an interrupt to tell it when the counts have been completed would be the way to do it.  The unfortunate part is that this will get very close to the correct frequency but I think there may still be some residual offset.

As for 10MHz to create a 38.4MHz clock, that is the easier route.  Silicon Labs has an [Si5338 EVB](http://www.silabs.com/products/clocksoscillators/clock-generator/Pages/Si5338-EVB.aspx) for $125 which has 8 outputs which could all be used to drive up to 8 different boards.  That's the same chip we use on the board itself, but they have other generators which have many more outputs for not much money.

polygon, 2015-08-07:
It's not possible to program the BladeRF Si5338 to use the 10 MHz reference for generating the 38.4 MHz?

bpadalino, 2015-09-16:
Sorry I didn't answer this before - it may be possible to use but it would change c4_clock to be 10MHz instead of 38.4MHz.  You would also want to make that clock output a pass-through instead of going through the internal PLL.  Then the c4_clock feeds the NIOS PLL.

There may be a way to do this directly using the si5338 on board, but with the FPGA controlling it - the brains might get scrambled before it can actually write the values unfortunately.

There was recently a patch to be able to provide 10MHz out the SMB port so you could lock other equipment to the bladeRF reference.

dkozel, 2015-09-17:
Being able to sync to an external 10 MHz reference would be very useful for phased array applications. I already have distribution ability for 10 MHz so it would be nice not to add another piece of hardware to the circuit just for the BladeRF.

I see that these have been on the todo list for a while. https://github.com/Nuand/bladeRF/wiki/Tasks

The RX sync on 1PPS is also a really nice feature. I certainly understand how you have a broad range of things you're working on. Are either of these on the roadmap for current development?

Thanks!

bpadalino, 2015-09-17:
Unfortunately it's lower on the priority list, but we'd be happy to give guidance for someone who wants to help contribute back into the project.

I understand about already having distribution and not wanting to add another piece of equipment.

The thing I worry about is getting exactly aligned versus being off by ~1Hz at 10MHz.  My guess if you want solid/stable locking and I am not sure a completely digital solution with a DAC would be able to give that.

thasti, 2015-09-19:
I'm working on this. Design is completed, some implementation already done. Expect a first sketch in a few days / weeks, depending on how much spare time I have. 

Some notes: Frequency *accuracy* is mainly dependent on the gate time used for counting the TCXO frequency. Example: When using 10 seconds, 0.1Hz resolution is possible. The maximum sensible gate time is limited by the short term *stability* of the TCXO. The optimal gate time can be found /by measuring the stability properties of the TCXO) and an achievable frequency accuracy can be found afterwards. I expect it to be better than 1Hz, but measurements will tell the truth.

What I think is not (easily) possible is implementing a digital PLL with the given hardware, which would be nice for locking multiple bladeRFs to the same reference for uses such as phased array applications, MIMO, ... - for these, the solution would be using the SMB output of one board as the TX clock of other bladeRFs.

jynik, 2015-12-14:
Marking this closed, as VCTCXO taming via 1PPS or 10MHz inputs (1.8V) via J71-1 has been introduced in FPGA v0.5.0, integrated into the libbladeRF API, and made available in the bladeRF-cli as of f6c0d28e6e4c872229cadf9c4373ea21ff933d8d.

closed 2015-12-14 by jynik

dkozel, 2015-12-14:
Awesome news

Keskebeu, 2018-04-18:
So in newer fpga's taming to 10mhz is still integrated to vctcxo?