Nuand/bladeRF #418: Add CMake project config files

jynik, 2015-08-15:
[CMake project config files](http://www.cmake.org/Wiki/CMake/Tutorials/How_to_create_a_ProjectConfig.cmake_file) help users of libbladeRF to locate it at configuration time via CMake.

We should be neighborly and provide these.

Examples from the [SoapySDR](https://github.com/pothosware/SoapySDR) Project:
* [SoapySDR *Config.cmake items](https://github.com/pothosware/SoapySDR/tree/9d9844d215222bd8e81bdf7557421feeffb3f997/cmake)
* [Installation of the above files](https://github.com/pothosware/SoapySDR/blob/dd50b2fc0d4f27396873c48270cf438c818dca2b/CMakeLists.txt#L46)

polygon, 2015-08-24:
I made something like that a while ago. Maybe it helps as a starter. https://gist.github.com/polygon/61e33a0fccecaf4ecf7a

Usage as normal via

```
find_package(LibbladeRF)
if (NOT LIBBLADERF_FOUND)
  message("libbladeRF not found")
else ()
  message(STATUS "Using libbladeRF from " ${LIBBLADERF_INCLUDE_DIRS})
  include_directories(${LIBBLADERF_INCLUDE_DIRS})
endif ()
```

jynik, 2016-05-25:
@guruofquality At a minimum, I'd like to install a FindLibbladeRF.cmake, if you'd allow for me to pull that into the project from SoapyBladeRF -- as you noted way back in, these are things we should really be providing to library users.

guruofquality, 2016-05-25:
@jynik sure you can knab the file https://github.com/pothosware/SoapyBladeRF/blob/master/FindLibbladeRF.cmake

But if you are going to do the cmake project files, we can totally drop the FindLibbladeRF.cmake and having people copy it into their projects. I might suggest a more cut-down project to copy from than SoapySDR: 

**muparserx**:

* https://github.com/beltoforion/muparserx/tree/master/cmake
* https://github.com/beltoforion/muparserx/blob/master/CMakeLists.txt#L87

**spuce**:

* https://github.com/audiofilter/spuce/tree/master/cmake
* https://github.com/audiofilter/spuce/blob/master/CMakeLists.txt#L55

Theres usually more than one way to do this, but the above stuff is pretty formulaic, 2 short files, and its worked so far across a wide range of cmakes going back to whatever is on 12.04 ubuntu.

cwozny, 2024-01-11:
Any ability for Nuand to add this capability? Seems like a light enough lift. Would love to just be able to have CMake find the package for me.

commit 01273a1a (2024-01-18): libbladeRF: added cmake package_config() support

Fixes #418
commit e4959463 (2024-01-18): libbladeRF: added cmake find_package() support

Fixes #418
commit 4ec99ebd (2024-01-18): libbladeRF: added cmake find_package() support

Fixes #418