Nuand/bladeRF #429: Duplex transmission don't work well with bladeRF-cli

khoikool, 2015-09-25:
I am doing research to use bladeRF-cli to transfer and receive data in full-duplex mode, I transmit a csv file and at the same time, I receive data and write it back to other csv file, like the tutorial in this website: 
https://github.com/Nuand/bladeRF/wiki/Getting-Started%3A-Verifying-Basic-Device-Operation
I find that these data I receive look so different from those data I transmitted. I don't know how to debug this problem. 
My code in -cli is:
bladeRF> rx config file=receive.csv format=csv n=10M
bladeRF> tx config file=transmit.csv format=csv repeat=10
bladeRF> tx start
Converted CSV to SC16 Q11 file and switched to converted file.
bladeRF> rx start
bladeRF> rx
    State: Idle
    Last error: None
    File: receive.csv
    File format: SC16 Q11, CSV
    # Samples: 10485760
    # Buffers: 32
    # Samples per buffer: 32768
    # Transfers: 16
    Timeout (ms): 1000
bladeRF> tx
    State: Running
    Last error: None
    File: bladeRF_samples_from_csv.bin
    File format: SC16 Q11, Binary
    Repetitions: 10
    Repetition delay: none
    # Buffers: 32
    # Samples per buffer: 32768
    # Transfers: 16
    Timeout (ms): 1000

bpadalino, 2015-09-25:
Can you explain how it looks so different?  Are you using internal or external loopback with attenuation?  Are your gains set appropriately?  Are you seeing a phase shift and you aren't sure how that looks?

jynik, 2015-09-26:
@khoikool, I'm going to ask that we take your answers to @bpadalino's questions and the remainder of this discussion to the [Nuand forums](https://www.nuand.com/forums), rather than discuss here on the issue tracker.  (If you are waiting for a forum account activation, please feel free to remind us via bladeRF@nuand.com)

The forum is a better place to share some screenshots and your sample files, and I think will overall get more exposure, which will help other people. If and when we can narrow down the scope of any particular problem or defect, we can open a tracker item here to track it.

jynik, 2015-09-26:
@khoikool  If the above is okay with you, please paste a link to the forum thread once we start discussing there, and I'll mark this closed for now.

khoikool, 2015-09-26:
@bpadalino , I transmit a stream of data ,>> I = 800, Q= 800 << in decimal, using antenna, I don't use loopback mode, I have setting gain, bandwidth, frequency. I receive >> I = -790 and Q= - 680.<<
@jynik I am kind of new to this website and don't know how to ask for support, if it's not convinient to discuss here, I will close this issue here and ask the Nuand forum for  further information, thank for your responds. :)

jynik, 2015-09-26:
@khoikool,

No worries - everyone is new at some point.  Where do you think it would it have been helpful for us to state that we prefer to offer this kind of support on the forums and IRC, and reserve this tracker for items that have been confirmed to be issues/defects?

Keep in mind that you're dealing with an analog signal and all the non-idealisties associated with analog hardware and the real world.  After data goes out the DAC, you will not not receive the exact same IQ values, due to differences in gain, analog offsets, etc.  @bpadalino and I can dive into more details on the forum, and hopefully post some examples to help you.

I will close this for now and will be sure we help answer your questions on the forum!

closed 2015-09-26 by jynik