Nuand/bladeRF #432: Error building on Pi

bnt2025, 2015-10-02:
On a fresh build of Rasbian on a Pi B+

I cannot install from the repository due to the pi having an old version of lib6c

pi@eink ~/bladeRF/host/build $ cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DINSTALL_UDEV_RULES=ON ../

[ 47%] Building C object libraries/libbladeRF_test/test_peripheral_timing/CMakeFiles/libbladeRF_test_peripheral_timing.dir/__/__/__/common/src/conversions.c.o
Linking C executable ../../../output/libbladeRF_test_peripheral_timing
/usr/bin/ld: CMakeFiles/libbladeRF_test_peripheral_timing.dir/main.c.o: undefined reference to symbol 'clock_gettime@@GLIBC_2.4'
//lib/arm-linux-gnueabihf/librt.so.1: error adding symbols: DSO missing from command line
collect2: ld returned 1 exit status
libraries/libbladeRF_test/test_peripheral_timing/CMakeFiles/libbladeRF_test_peripheral_timing.dir/build.make:137: recipe for target 'output/libbladeRF_test_peripheral_timing' failed
make[2]: *** [output/libbladeRF_test_peripheral_timing] Error 1
CMakeFiles/Makefile2:654: recipe for target 'libraries/libbladeRF_test/test_peripheral_timing/CMakeFiles/libbladeRF_test_peripheral_timing.dir/all' failed
make[1]: *** [libraries/libbladeRF_test/test_peripheral_timing/CMakeFiles/libbladeRF_test_peripheral_timing.dir/all] Error 2
Makefile:113: recipe for target 'all' failed
make: *** [all] Error 2

bnt2025, 2015-10-03:
I tried building on the new Raspbian Jesse image on the Raspberry pi website with kernel 4.1 and it built successfully.

closed 2015-10-03 by bnt2025

jynik, 2015-10-03:
Do you still have the older image to test with? I'd like to understand why the build was failing.  If not, could you let me know what the previous image was so I can look into this?

[This line](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF_test/test_peripheral_timing/CMakeLists.txt#L26) should have ensured that with that older libc version you had libbladeRF_test_peripheral_timing would be  built with `-lrt`.  I'm curious as to why this was not the case.

If possible, I'd like to see the output for:
```
$ cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DINSTALL_UDEV_RULES=ON ../
$ make VERBOSE=1
```

If you could

dballenger, 2016-05-14:
I just ran into this, attaching the logs if y
[build.txt](https://github.com/Nuand/bladeRF/files/264498/build.txt)

ou're still curious.

jynik, 2016-05-18:
@DanielG Thanks for the full log -- definitely good to have this in the issue history.

Were you able to get things working on a newer RPi image?

dballenger, 2016-05-18:
Yeah, it built fine on an updated image that flightaware provides.

jynik, 2016-05-19:
I'm a little unclear as to history of the ARM `libc` and `librt` on these images, with respect to the move of `clock_gettime()` and whether this is only associated with the RPi images or other platforms.

Given that the images that yield this build failure also contain a libusb version that is vastly older than what we'd recommend using, I'm going to treat this as a `wontfix` situation.