Nuand/bladeRF #434: Compile error on v1.4.3 tag

staticfloat, 2015-10-08:
From a fresh clone on an OSX 10.10 machine:

```
[  1%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/async.c.o
In file included from /Users/sabae/src/gnuradio/bladeRF/host/libraries/libbladeRF/src/async.c:24:
In file included from /Users/sabae/src/gnuradio/bladeRF/host/libraries/libbladeRF/src/async.h:26:
/Users/sabae/src/gnuradio/bladeRF/host/libraries/libbladeRF/src/bladerf_priv.h:139:5: error: unknown type name
      'bladerf_tuning_mode'; did you mean 'bladerf_lpf_mode'?
    bladerf_tuning_mode tuning_mode;
    ^~~~~~~~~~~~~~~~~~~
    bladerf_lpf_mode
/Users/sabae/local/include/libbladeRF.h:517:3: note: 'bladerf_lpf_mode' declared here
} bladerf_lpf_mode;
  ^
1 error generated.
```

Here is the [full `cmake` and `make` output](https://gist.github.com/staticfloat/630b23304ce547a1a6b1).  This is on v1.4.3, (60714d7f61f5783bb82d3b14d1b9d742334d4769) with no local modifications.

staticfloat, 2015-10-08:
Ah, it looks like this was due to old bladerf headers laying around in my prefix.

closed 2015-10-08 by staticfloat

jynik, 2015-10-08:
@staticfloat Good catch!