Nuand/bladeRF #441: Installing bladeRF-cli removes gqrx

mitos14, 2015-12-22:
Hi,
I had gqrx installed, then I installed bladeRF-cli from your ppa and when I type sudo gqrx on the commandline I get program not found. So I did: apt-get install gqrx-sdr, which worked, but then bladeRF-cli was now gone (meaning I can not run it from the command line).
Am I doing something wrong here or is this an issue?
thank you

jynik, 2015-12-23:
* What distro are you on?
 * What are the dependencies of the gqrx-sdr package?
  * Does it depend on an old bladeRF package version that conflicts with what you're trying to install from the PPA?

* Why are you running GQRX as root?
 * If you have a proper udev rule installed, this should not be necessary

jynik, 2015-12-29:
@mitos14 

Until we can confirm that there's a defect here that's within the realm of something we can fix, let's discuss this elsewhere (e.g., the Nuand forums).

closed 2015-12-29 by jynik