Nuand/bladeRF #443: [libbladeRF] Cypress backend receiving NULL buffers upon transfer completion

jynik, 2015-12-29:
During testing in Windows with the CyUSB3 driver, I've been experiencing problems with the TX stream receiving transfer completions from the CyUSB3 driver consisting of NULL buffers.

To work around this for the time being, the libusb backend may be used instead.

jynik, 2016-06-17:
I've worked back through libbladeRF v1.4.0 and still was able to reproduce this issue.  Based upon my previous testing, these versions were passing tests that now fail.

It may be the case that this isn't a libbladeRF regression, but rather related to the CyFX3 SDK 1.3.1 -> 1.3.3 upgrade (includes Cypress driver and library updates).  Will update this issue once I've tried rolling those back.

jynik, 2016-06-17:
Reverted to CyFX3 SDK 1.3.1 and still reproduced this failure.

rtucker, 2017-08-07:
Reproduction currently blocked on #523