Nuand/bladeRF #447: Build broken on MSVC2015

jynik, 2016-01-11:


jynik, 2016-01-11:
Closed by a459c212096e1a3a4f3f101b09d8acb49163d217

jynik, 2016-06-29:
Re-opened this issue. It appears that VS2015 adds a `struct timespec` implementation that is provided by the Windows pthreads implementation.

This breaks the build anywhere that the Windows `pthreads.h` implementation is included.

jynik, 2016-06-29:
More information leading to potential fix:

http://mingw-users.1079350.n2.nabble.com/mingw-error-redefinition-of-struct-timespec-td7583722.html

jynik, 2016-07-12:
@guruofquality I see there's some discussion in the MinGW world (linked above) about what I assume are ABI concerns WRT to the MS timespec and the pthreadswin32 version.  (i.e., the need to rebuild pthreads with `HAVE_STRUCT_TIMESPEC`) 

Did you find that you had to rebuild?

guruofquality, 2016-07-12:
Just for reference, its been working but the bug probably manifests itself as a very large timeout. Which is probably ok in some situations. I made an issue to fix this in my [PothosSDR installer](https://github.com/pothosware/PothosSDR/issues/27). Thanks for pointing that out.

rtucker, 2017-08-04:
Faking it by editing `pthreads.h` is plausible, but the ABI concerns are valid.  So, I am adding the following rebuild process to host/README.md.  I am pondering whether we want to test for this situation during CMake, or if I should just hold out hope that it'll get fixed in pthreads-win32 itself at some point...

#### pthreads-win32 ####
If you are using a newer version of Windows / Visual Studio and are getting ```error C0211: 'timespec': 'struct' type redefinition``` when trying to build bladeRF, you will need to rebuild pthreads-win32. The below process seems to work with pthreads-w32 2.9.1.

- Get full source tree from [pthreads-win32] website, and unpack someplace convenient
- In Visual Studio, open `pthreads-w32-2-9-1-release\pthreads.2\pthread.dsw` (one-way upgrade is OK)
- Right-click the solution, select "Retarget solution", hit OK.
- Change "Solution Configurations" dropdown to "Release" ("Debug" doesn't work, but that's OK)
- Edit pthread.h to add `#define HAVE_STRUCT_TIMESPEC` and `#define PTW32_ARCHx64` near the top, just below the include guard
- Build the solution

- Back over in the bladeRF CMake:
    - File -> Delete Cache
    - Redo the "Configure" dance as before
    - Point `LIBPTHREADSWIN32_PATH` to the `pthreads-w32-2-9-1-release\pthreads.2` directory you were just working in

rtucker, 2017-08-04:
note to self: update https://github.com/Nuand/bladeRF/wiki/Getting-Started:-Windows

closed 2017-08-07 by rtucker