Nuand/bladeRF #453: Make docs clear that cal dc doesn't actually apply values if they aren't provided.

johnmoe1, 2016-02-09:
I ran:
bladeRF> cal dc rx

RX DC I: Value =   384, Error =  0.500
RX DC Q: Value =   480, Error =  2.702

expecting that the average I value would be close to zero. It was actually about -70 and I was confused.

I then ran:
bladeRF> cal dc rx 384 480

the average I value was now about zero.

jynik, 2016-02-09:
It is intended to apply the values found from searching. I would need to
seem some samples from you to better understand what you're seeing.

johnmoe1, 2016-02-09:
It doesn't seem to. Does this help?

$ bladeRF-cli -i -v verbose
[VERBOSE @ libusb.c:536] Using libusb version: 1.0.17.10830
[VERBOSE @ libusb.c:417] Found a bladeRF (idx=4)
[VERBOSE @ libusb.c:479] USB port reset succeeded for bladeRF 73fbc8a98f380b28cec4a81896a4682d
[VERBOSE @ capabilities.c:45] Capability mask before FPGA load: 0x0000000300000000
[VERBOSE @ usb.c:112] Changing to USB alt setting 0
[VERBOSE @ usb.c:112] Changing to USB alt setting 1
[VERBOSE @ usb.c:169] Read FPGA version: 0.5.0
[VERBOSE @ capabilities.c:88] Capability mask after FPGA load: 0x00000003000001ff
[VERBOSE @ usb.c:180] Using current packet handler formats
[VERBOSE @ usb.c:112] Changing to USB alt setting 2
[VERBOSE @ usb.c:112] Changing to USB alt setting 1
[VERBOSE @ usb.c:112] Changing to USB alt setting 2
[VERBOSE @ usb.c:112] Changing to USB alt setting 1
[VERBOSE @ nios_access.c:723] Clearing RX retune queue.
[VERBOSE @ nios_access.c:746] RX operation duration: 0
[VERBOSE @ nios_access.c:723] Clearing TX retune queue.
[VERBOSE @ nios_access.c:746] TX operation duration: 0
[VERBOSE @ nios_access.c:298] nios_config_read: Read 0x0000004f
[DEBUG @ tuning.c:82] Default tuning mode: FPGA
[VERBOSE @ nios_access.c:298] nios_config_read: Read 0x0000004f
bladeRF> cal dc rx
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x59 from addr 0x20
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x02 from addr 0x21
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0xb5 from addr 0x22
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x54 from addr 0x23
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0xf9 from addr 0x25
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x9d from addr 0x29
[VERBOSE @ nios_access.c:298] nios_config_read: Read 0x0000004f
[VERBOSE @ si5338.c:288] Reading MS1
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x01 from addr 0x25
[VERBOSE @ si5338.c:297] Read enable register: 0x01
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x27 from addr 0x40
[VERBOSE @ si5338.c:306] Read regs[0]: 0x27
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x17 from addr 0x41
[VERBOSE @ si5338.c:306] Read regs[1]: 0x17
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x08 from addr 0x42
[VERBOSE @ si5338.c:306] Read regs[2]: 0x08
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x00 from addr 0x43
[VERBOSE @ si5338.c:306] Read regs[3]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x00 from addr 0x44
[VERBOSE @ si5338.c:306] Read regs[4]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x00 from addr 0x45
[VERBOSE @ si5338.c:306] Read regs[5]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x52 from addr 0x46
[VERBOSE @ si5338.c:306] Read regs[6]: 0x52
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x00 from addr 0x47
[VERBOSE @ si5338.c:306] Read regs[7]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x00 from addr 0x48
[VERBOSE @ si5338.c:306] Read regs[8]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0x00 from addr 0x49
[VERBOSE @ si5338.c:306] Read regs[9]: 0x00
[VERBOSE @ nios_access.c:385] nios_si5338_read: Read 0xc0 from addr 0x20
[VERBOSE @ si5338.c:316] Read r register: 0xc0
[VERBOSE @ si5338.c:156] Unpacked P1: 0x00001727 (5927) P2: 0x00000002 (2) P3: 0x00000052 (82)
[VERBOSE @ si5338.c:174] Unpacked a + b/c: 50 + 25/82
[VERBOSE @ si5338.c:175] Unpacked r: 1
[VERBOSE @ si5338.c:347] Calculated multisynth frequency: 25190400 + 0/1
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x0e from addr 0x54
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x3f from addr 0x10
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0xb9 from addr 0x11
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x55 from addr 0x12
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x55 from addr 0x13
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0xb2 from addr 0x15
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x9e from addr 0x19
[VERBOSE @ nios_access.c:298] nios_config_read: Read 0x0000004f
[VERBOSE @ si5338.c:525] Setting integer sample rate: 3000000
[VERBOSE @ si5338.c:383] Found r value of: 1
[... 212 lines cut ...]
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x3e to addr 0x54
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x19 from addr 0x55
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x3e from addr 0x54
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x0e to addr 0x54
[DEBUG @ tuning.c:134] Setting TX frequency to 2446999999
[VERBOSE @ lms.c:357] VCOCAP estimate: 31
[VERBOSE @ lms.c:1677] ---- Frequency ----
[VERBOSE @ lms.c:1678]   x        : 2
[VERBOSE @ lms.c:1679]   nint     : 127
[VERBOSE @ lms.c:1680]   nfrac    : 3757397
[VERBOSE @ lms.c:1681]   freqsel  : 0x2c
[VERBOSE @ lms.c:1682]   reference: 38400000
[VERBOSE @ lms.c:1683]   freq     : 2446999999
[VERBOSE @ nios_access.c:728] nios_retune: module=TX timestamp=0 nint=127 nfrac=3757397
				freqsel=0x2c vcocap=0x1f low_band=0 quick_tune=0
[VERBOSE @ nios_access.c:743] TX retune operation: vcocap=30, duration=382

RX DC I: Value =   384, Error =  0.332
RX DC Q: Value =   512, Error =  2.670

bladeRF> cal dc rx 384 512
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x9b from addr 0x71
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x8c to addr 0x71
[VERBOSE @ nios_access.c:409] nios_lms6_read: Read 0x9b from addr 0x72
[VERBOSE @ nios_access.c:421] nios_lms6_write: Wrote 0x90 to addr 0x72

jynik, 2016-02-09:
Did you run 'cal lms' before this, by the way?

jynik, 2016-02-09:
I was able to reproduce this. It seems like the nominal correction values may be getting reverted or not restored after they are evaluated.

This command is absolutely intended to apply the value it finds, so I am changing the title of this issue.

I will investigate further and let you know what I find.  Thanks for reporting this!

jynik, 2016-02-10:
I pushed a fix to this in 7c55150c9865fa7d79f825269909291d065e279e.  The nominal values simply weren't being applied. The values in the LMS6 registers were just the last tested values.

closed 2016-02-10 by jynik