Nuand/bladeRF #454: [debian] bladerf-fpga-hostedx40 leaves bad rbf file around if the download fails

rtucker, 2016-02-13:
Third party report:

Hi Ryan,
Went through the process of getting the BladeRF running again today,
and ran into a funky issue with the PPA.

TL;DR:

I'm using Linux Mint 17.3 - KDE Spin. I setup the PPA and installed
all the things:
```
$ sudo add-apt-repository ppa:bladerf/bladerf
$ sudo apt-get update
$ sudo apt-get install bladerf libbladerf-dev bladerf-firmware-fx3
bladerf-fpga-hostedx40
```

The install of the bladerf-fpga-hostedx40 package failed with an error
complaining about being unable to connect to localhost port 8080. In
the output of that failure it told me to run dpkg-reconfigure, so I
did that, with the same result:

```
$ sudo dpkg-reconfigure bladerf-fpga-hostedx40
Downloading FPGA bitstream from nuand.com...
--2016-02-13 10:52:52--  https://www.nuand.com/fpga/v0.5.0/hostedx40.rbf
Connecting to 127.0.0.1:8080... failed: Connection refused.
Warning: Failed to download FPGA bitstream for bladeRF.
Please run "dpkg-reconfigure bladerf-fpga-hostedx40" again when
networking is up, or copy the firmware manually to
/usr/share/Nuand/bladeRF/hostedx40.rbf
```

====

The main problem with this is that a broken fpga image is left in
/usr/share/Nuand/bladeRF/hostedx40.rbf. The way this manifests in the
system is like so:

```
$ bladeRF-cli -p

  Backend:        libusb
  Serial:         d1c8b59ff5d36d39d43af24eaca005a3
  USB Bus:        2
  USB Address:    4

$ bladeRF-cli -i
Failed to open device (first available): Invalid operation or parameter

$ groups
jross adm dialout cdrom sudo dip plugdev lpadmin sambashare

$ bladeRF-cli -i -v verbose
[VERBOSE @ libusb.c:536] Using libusb version: 1.0.17.10830
[VERBOSE @ libusb.c:417] Found a bladeRF (idx=4)
[VERBOSE @ libusb.c:479] USB port reset succeeded for bladeRF
d1c8b59ff5d36d39d43af24eaca005a3
[VERBOSE @ capabilities.c:45] Capability mask before FPGA load:
0x0000000300000000
[VERBOSE @ usb.c:112] Changing to USB alt setting 0
[VERBOSE @ usb.c:112] Changing to USB alt setting 2
[VERBOSE @ usb.c:112] Changing to USB alt setting 3
[VERBOSE @ usb.c:112] Changing to USB alt setting 2
[VERBOSE @ usb.c:112] Changing to USB alt setting 3
[DEBUG @ config.c:80] Loading FPGA from:
/home/jross/.Nuand/bladeRF/hostedx40.rbf
Failed to open device (first available): Invalid operation or parameter
```

That is what made me think something was up with the FPGA image. The
hostedx40.rbf file it's opening there is just a symlink to the one in
/usr/share/Nuand/bladeRF/hostedx40.rb. To fix this, I simply did this:

```
# cd /usr/share/Nuand/bladeRF
# wget https://www.nuand.com/fpga/v0.5.0/hostedx40.rbf
```

That did it:

```
$ bladeRF-cli -i
bladeRF> info

  Serial #:                 d1c8b59ff5d36d39d43af24eaca005a3
  VCTCXO DAC calibration:   0x8ea1
  FPGA size:                40 KLE
  FPGA loaded:              yes
  USB bus:                  2
  USB address:              4
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0
```

rtucker, 2016-02-13:
My response:

Oh that's weird... looks like, for whatever reason, wget was trying to
use a proxy on localhost:8080 or something, and it failed.  This is
likely something weird in your environment, and I probably can't fix
that.  HOWEVER, it shouldn't leave a bad file there.

I'll make sure it cleans up after itself and will advise.  -rt

rtucker, 2016-02-13:
Pull request #455 created for a fix on this issue

rtucker, 2016-02-13:
Dockerfile to reproduce issue: https://gist.github.com/rtucker/3d3f07b79afba9da10a8

rtucker, 2016-02-13:
Couldn't reproduce the "Connecting to 127.0.0.1:8080... failed: Connection refused." issue with a stock Linux Mint 17.3 KDE install.

jynik, 2016-02-14:
Addressed by merge of PR #455.

@rtucker - many thanks for the quick PR response and detailed report. Cheers! :beers:

closed 2016-02-14 by jynik