Nuand/bladeRF #458: [libbladerf] assertion failed in sync_worker.c, tx_callback after bug 438 fix

jquirke, 2016-02-23:
Since upgrading to libbladerf 2016.01-rc1 on the Cygwin64 platform I have received the following assertion failure in tx_callback 3 times in 2 days of using the lib:

assert(b->status[completed_idx] == SYNC_BUFFER_IN_FLIGHT);

In each case, the b->status[completed_idx] is 

` p b->status[completed_idx]`
`$5 = SYNC_BUFFER_FULL`

and the status of all my buffers (13 buffers, 12 transfers, 2048 samples/buf) is

`(gdb) p *b->status@13`
`$2 = {SYNC_BUFFER_EMPTY, SYNC_BUFFER_EMPTY, SYNC_BUFFER_EMPTY, SYNC_BUFFER_EMPTY, SYNC_BUFFER_EMPTY,
  SYNC_BUFFER_EMPTY, SYNC_BUFFER_EMPTY, SYNC_BUFFER_FULL, SYNC_BUFFER_EMPTY, SYNC_BUFFER_EMPTY, SYNC_BUFFER_EMPTY,
  SYNC_BUFFER_EMPTY, SYNC_BUFFER_EMPTY}`
`(gdb) p *b`
`$3 = {status = 0x60019cd00, buffers = 0x60018a240, num_buffers = 13, prod_i = 7, cons_i = 4294967295,
  partial_off = 0, resubmit_count = 0, submitter = SYNC_TX_SUBMITTER_FN, lock = 0x600164e30, buf_ready = 0x60018aa00}`
`(gdb) p completed_idx`
`$4 = 7`
`

I can reproduce it by waiting long enough, it happens randomly. What info would be useful?

jquirke, 2016-02-23:
I believe this code was reworked to fix bug 438, where the sync_worker was not resubmitting tx buffers in its own context...

I also noticed this problem last year and hence why I am using 12 transfers, 13 buffers (there was no point using any more under the old version).

jquirke, 2016-02-23:
It's late but my hypothesis is a race condition in advance_tx_buffer during which time b->lock is unlocked, a buffer can submit and complete (with status SYNC_BUFFER_FULL) *before* the advance_tx_buffer has finished.

Quite plausible with my small buffers?

I'll see if I can prove this.

jynik, 2016-02-23:
@jquirke Thanks for reporting this. I'd appreciate any additional info you can come up with or ways to reproduce this.

I haven't reviewed enough to make any claims, but I'm wondering if that changes you suspect are also the same responsible for #443....

jquirke, 2016-02-24:
I have a "repro", which is to insert a short pause before acquiring the lock in advance_tx_buffer

In my example, use buffersize = 2048 samples, xfers=12, buffers=13

Add a 250usec pause (simulate CPU scheduling etc), and it blows up the first time

`status = async_submit_stream_buffer(s->worker->stream,
                                            b->buffers[idx],
                                            s->stream_config.timeout_ms,
                                            true);`
											
`LARGE_INTEGER time1,time2,freq;`
		`QueryPerformanceCounter((LARGE_INTEGER *) &time1);`
		`QueryPerformanceFrequency((LARGE_INTEGER *)&freq);`

		`do {`
			`QueryPerformanceCounter((LARGE_INTEGER *) &time2);`
		`} while((time2.QuadPart-time1.QuadPart) < (freq.QuadPart/1000/4));`
		
        `MUTEX_LOCK(&b->lock);`

Anyway, I don't believe the lock needs to be acquired anymore since the async_submit_stream_buffer is used in non-blocking mode. This just creates a race condition.

So I have removed it and all good. This solves the problem for me, so this issue can be closed.

jynik, 2016-02-25:
@jquirke Thanks - I'll try to follow this to reproduce it and look into it further.  Keep me in the loop if you dig any further on this.

jynik, 2016-02-25:
@jquirke I haven't gotten this to blow up on me yet, but I agree with your assessment and believe I have a patch that addresses this race.

Would you mind taking a look at and trying out the patch in the [dev-issue_458](https://github.com/Nuand/bladeRF/tree/dev-issue_458) branch? Thanks!

jquirke, 2016-02-26:
@jynik Thanks; really appreciated; I will give it a try over the weekend.

For my quasi-repro, I should probably also mention my samplerate was 7.68MHz. This means 250usec delay was roughly on the order of a buffer. Otherwise the delay should be adjusted accordingly.

As for the frequency of this, it's happened another half dozen or so times; so I'll know very quickly if the patch works.

jquirke, 2016-02-27:
@jynik the dev-issue_458 branch works a treat.

Had it running for hours, no repro. I found before the patch I could repro every time by dragging windows around rapidly, now it never happens.

jynik, 2016-02-27:
@jquirke Thanks for the quick confirmation and your analysis of the issue - it's much appreciated!

Fix merged in ba561a4aa1c5aa949b07cf55e333d626c8b98cf3.

commit ba561a4a (2016-02-27): libbladeRF: Fixed race condition in TX buffer submission

It is possible that the buffer completion callback occurs before
advance_tx_buffer() acts upon a successful return code from
async_submit_stream_buffer().

We must ensure the buffer is in the IN_FLIGHT state before
we temporarily drop the buffer lock (so a stream lock can be acquired),
or else the assertion at sync_worker.c:150 may not be upheld.

Therefore, we default the buffer state to being IN_FLIGHT before
dropping the lock, and revert it to FULL if the buffer submission
fails (due to an error or a need to block in a non-blocking c [... cut]
closed 2016-02-27 by jynik