Nuand/bladeRF #459: [libbladeRF] Build failure reported on MS VS2013

jynik, 2016-02-23:
I have not yet been able to reproduce this, but below is a log of a reported build failure on Visual Studio 2013 (Update 4).  We pull in the timespec definition courtesy of the pthreads win32 library, so I'm unclear as to why timespec would be undefined...

```
1>------ Build started: Project: libbladerf_shared, Configuration: Debug Win32 ------
2>------ Build started: Project: win_runtime, Configuration: Debug Win32 ------
3>------ Skipped Build: Project: uninstall, Configuration: Debug Win32 ------
3>Project not selected to build for this solution configuration 
1>Build started 2/23/2016 2:27:16 PM.
2>Build started 2/23/2016 2:27:16 PM.
2>InitializeBuildStatus:
2>  Touching "Win32\Debug\win_runtime\win_runtime.tlog\unsuccessfulbuild".
1>InitializeBuildStatus:
1>  Touching "libbladerf_shared.dir\Debug\libblade.6224E286.tlog\unsuccessfulbuild".
1>CustomBuild:
1>  All outputs are up-to-date.
2>CustomBuild:
2>  Building Custom Rule F:/projects/blade_build/bladeRF/host/CMakeLists.txt
2>  CMake does not need to re-run because F:\projects\blade_build\build_dbg\CMakeFiles\generate.stamp is up-to-date.
2>  Copying pthreads-win32 license to output directory...
2>  Copying pthreadVC2.dll to output directory...
2>  Copying libusb license to output directory...
1>ClCompile:
1>  async.c
2>  Copying libusb-1.0.dll to output directory...
2>  Copying Cypress License to output directory...
2>  Copying cyusb3.inf to output directory...
2>  Copying cyusb3.sys to output directory...
2>  Copying cyusb3.cat to output directory...
2>  Copying WdfCoInstaller01011.dll to output directory...
2>  Error copying file (if different) from "f:/fx3/EZ-USB_FX3_SDK/1.2/driver/bin/win7/x64/WdfCoInstaller01011.dll" to "F:/projects/blade_build/build_dbg/output/Debug/drivers/CyUSB3/".
2>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V120\Microsoft.CppCommon.targets(170,5): error MSB6006: "cmd.exe" exited with code 1.
2>
2>Build FAILED.
2>
2>Time Elapsed 00:00:00.27
1>F:\projects\blade_build\bladeRF\host\libraries\libbladeRF\src\async.c(160): error C2079: 'timeout_abs' uses undefined struct 'timespec'
1>F:\projects\blade_build\bladeRF\host\libraries\libbladeRF\src\async.c(166): warning C4133: 'function' : incompatible types - from 'int *' to 'timespec *'
1>F:\projects\blade_build\bladeRF\host\libraries\libbladeRF\src\async.c(182): warning C4133: 'function' : incompatible types - from 'int *' to 'const timespec *'
1>
1>Build FAILED.
1>
1>Time Elapsed 00:00:00.63
4>------ Build started: Project: bladeRF-cli, Configuration: Debug Win32 ------
5>------ Build started: Project: libbladeRF_test_async, Configuration: Debug Win32 ------
6>------ Build started: Project: libbladeRF_test_bootloader_recovery, Configuration: Debug Win32 ------
7>------ Build started: Project: libbladeRF_test_c, Configuration: Debug Win32 ------
8>------ Build started: Project: libbladeRF_test_cpp, Configuration: Debug Win32 ------
9>------ Build started: Project: libbladeRF_test_ctrl, Configuration: Debug Win32 ------
10>------ Build started: Project: libbladeRF_test_freq_hop, Configuration: Debug Win32 ------
11>------ Build started: Project: libbladeRF_test_open, Configuration: Debug Win32 ------
4>Build started 2/23/2016 2:27:17 PM.
5>Build started 2/23/2016 2:27:17 PM.
11>Build started 2/23/2016 2:27:17 PM.
5>PrepareForBuild:
5>  Creating directory "libbladeRF_test_async.dir\Debug\libblade.5CED1F62.tlog\".
4>PrepareForBuild:
4>  Creating directory "bladeRF-cli.dir\Debug\bladeRF-cli.tlog\".
4>InitializeBuildStatus:
4>  Creating "bladeRF-cli.dir\Debug\bladeRF-cli.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
5>InitializeBuildStatus:
5>  Creating "libbladeRF_test_async.dir\Debug\libblade.5CED1F62.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
6>Build started 2/23/2016 2:27:17 PM.
5>CustomBuild:
5>  Building Custom Rule F:/projects/blade_build/bladeRF/host/libraries/libbladeRF_test/test_async/CMakeLists.txt
4>CustomBuild:
4>  Building Custom Rule F:/projects/blade_build/bladeRF/host/utilities/bladeRF-cli/CMakeLists.txt
5>  CMake does not need to re-run because F:\projects\blade_build\build_dbg\libraries\libbladeRF_test\test_async\CMakeFiles\generate.stamp is up-to-date.
4>  CMake does not need to re-run because F:\projects\blade_build\build_dbg\utilities\bladeRF-cli\CMakeFiles\generate.stamp is up-to-date.
4>  Generating src/cmd/doc/cmd_help.h
9>Build started 2/23/2016 2:27:17 PM.
[... 409 lines cut ...]
4>F:\projects\blade_build\bladeRF\host\utilities\bladeRF-cli\src\cmd\rxtx.c(803): warning C4133: 'function' : incompatible types - from 'int *' to 'const timespec *'
4>
4>Build FAILED.
4>
4>Time Elapsed 00:00:13.84
22>------ Build started: Project: ALL_BUILD, Configuration: Debug Win32 ------
22>Build started 2/23/2016 2:27:31 PM.
22>PrepareForBuild:
22>  Creating directory "Win32\Debug\ALL_BUILD\ALL_BUILD.tlog\".
22>InitializeBuildStatus:
22>  Creating "Win32\Debug\ALL_BUILD\ALL_BUILD.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
22>CustomBuild:
22>  Building Custom Rule F:/projects/blade_build/bladeRF/host/CMakeLists.txt
22>  CMake does not need to re-run because F:\projects\blade_build\build_dbg\CMakeFiles\generate.stamp is up-to-date.
22>FinalizeBuildStatus:
22>  Deleting file "Win32\Debug\ALL_BUILD\ALL_BUILD.tlog\unsuccessfulbuild".
22>  Touching "Win32\Debug\ALL_BUILD\ALL_BUILD.tlog\ALL_BUILD.lastbuildstate".
22>
22>Build succeeded.
22>
22>Time Elapsed 00:00:00.20
23>------ Skipped Build: Project: INSTALL, Configuration: Debug Win32 ------
23>Project not selected to build for this solution configuration 
========== Build: 1 succeeded, 20 failed, 2 up-to-date, 2 skipped ==========
```

jynik, 2016-02-29:
Have not been able to reproduce this. I also see that the log indicates an unsupported FX3 SDK is installed.

Marking this `invalid` and closing this issue. Feel free to re-open this and add additional information, as needed.

closed 2016-02-29 by jynik

roflo1, 2019-11-06:
I was able to reproduce this "timespec" error in VS2013 Pro; however, the exact same project in VS2017 Community (in the same machine) compiles without much hassle.

The log above also mentions an error while copying "WdfCoInstaller01011.dll" to the output folder. This is because the file doesn't exist in the FX3 API folder for Win10. This didn't prevent me from building bladeRF-cli (and dependencies); and since I had already installed the device driver I ignored this (didn't test).