Nuand/bladeRF #462: cmake build failure on CentOS 7.2

mmilleror, 2016-03-07:
Trying to build on CentOS 7.2.  When cmake is run it errors out with the following.

cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr/local -DINSTALL_UDEV_RULES=ON ../
-- Check if the system is big endian
-- Searching 16 bit integer
CMake Error at /usr/share/cmake/Modules/TestBigEndian.cmake:44 (message):
  no suitable type found
Call Stack (most recent call first):
  host/CMakeLists.txt:65 (TEST_BIG_ENDIAN)

-- Configuring incomplete, errors occurred!

jynik, 2016-03-08:
It seems like this is failing in module shipped with CMake itself, rather
than the bladeRF codebase.

Are you able to reproduce this on any other CMake-based projects that have
the following line in their CMakeLists.txt?

include(TestBigEndian)

From what you've shown, it sounds like this might be a defect in CMake or
the CentOS packaging, rather than this projects. Let's try to determine
whether or not my assessment here is correct.

mmilleror, 2016-03-08:
I would agree that it's a issue with cmake.  If I comment out include(TestBigEndian) and TEST_BIG_ENDIAN(BLADERF_BIG_ENDIAN) cmake runs fine.  When I uncomment include(TestBigEndian) and leave the TEST_BIG_ENDIAN(BLADERF_BIG_ENDIAN) cmake runs fine.  I was able to build bladeRF just fine.  I thought it might be an issue with the BLADERF_BIG_ENDIAN variable. Maybe not...  I'll take a look at the cmake change log to see if anything was fixed in 2.8.12.

jynik, 2016-03-09:
@mmilleror Shall I mark this closed then?  It sounds like there's not much we can do on our end.

mmilleror, 2016-03-09:
I guess.  It builds find on Fedora 23.  I just didn't know where the breakdown was with cmake are when you set the variable.

jynik, 2016-03-22:
@mmilleror Marking this closed per our discussion of this seemingly being more of an issue with some CMake files.  

Please do feel free to continue commenting here.  I'd definitely like to have whatever you find out to be on the record here for other users.

closed 2016-03-22 by jynik