Nuand/bladeRF #463: Adding SignalTap in scripted build

Konstantin41, 2016-03-08:
Trying to follow the instructions how to build FPGA firmware using scripted build  https://github.com/Nuand/bladeRF/tree/master/hdl,this is not working!!!,better to use GUI,open the Quartus project in the GUI, verified SignalTap is enabled and using a valid .stp file, then re-run Analysis & Synthesis, Fitter, and Generate Programming File manually using the GUI.

bpadalino, 2016-03-08:
The procedure you follow is about the best we can do.  For some reason it doesn't get included on the first time the script builds even though, as you noted, the project file settings are all correct.

I believe this to be a Quartus bug.  If you want to reduce time for compilation, run the script to create the project, exit out of the build, then run with the GUI.  That seems to work.

jynik, 2016-03-09:
Marking this "wontfix" because we've reportedly confirmed that it's a quartus bug.

If anyone can confirm this to only be a build script bug, we can re-open this issue and go from there.

jynik, 2016-03-09:
@Konstantin41 - saw your comment on the forums.

Yes, we should figure out where it would be helpful to let folks know that they can't do this from the command line. 

Where would you have noticed such a note? Would placing it in the HDL README have been a place you would have seen it?

Konstantin41, 2016-03-09:
Yes,it would be good,if you will add this instructions in HDL README in SignalTap section

bglod, 2016-04-02:
Fixed in commit 1f0e6f65e06549e16fcb1afa886922f2bf813286.

closed 2016-04-02 by bglod