Nuand/bladeRF #468: BLADERF_FORMAT_SC16_Q11_META docs confusing

johnmoe1, 2016-05-01:
>     /**
     * This format is the same as the ::BLADERF_FORMAT_SC16_Q11 format, except the
     * first 4 samples (16 bytes) in every block of 1024 samples are replaced
     * with metadata, organized as follows, with all fields being little endian
     * byte order:

Is a sample 2 bytes or 4 bytes? This sentence is confusing. Saying either "first 8 samples (16 bytes) in every block of 1024 samples (2048 bytes)" or "first 4 samples (16 bytes) in every block of 512 samples (2048 bytes)" seems less confusing to me. Or maybe just stick with bytes and forget about samples?

jynik, 2016-05-01:
Hi @johnmoe1 - let me answer your question first, and then see what we can come up with together to make the documentation more clear.

Samples are 4 bytes:
* A sample consists of I followed by Q
* I and Q are int16_t values (with valid ranges of [-2048, 2047])
* Thus, a sample = `2 * sizeof(int16_t) = 2 * 2 = 4`

Thus, we could change that to read:

```
...first 4 samples (16 bytes) in every block of 1024 samples (4096 bytes) are replaced...
```

The intent with listing both samples and bytes was simply to articulate how many samples the metadata "cost" you. (Perhaps that was not a good idea...)

Now, on top of what you've pointed out, I now realize that the `in every block of 1024 samples` is not technically correct and needs to be fixed.  If you take a look at [this file](https://github.com/Nuand/bladeRF/blob/master/host/libraries/libbladeRF/src/metadata.h#L21) you'll see the relationship between this metadata and the samples that follow.   The whole scheme with "messages" and "buffers" is a function of FPGA <-> FX3 <-> Host transfer (e.g., DMA) lengths.

It should *actually* read `256 samples (USB 2.0) or 512 samples (USB 3.0)`!

I know I've tossed a lot at you, but would definitely appreciate an another pair of eyes assessing the readability of the docs.

johnmoe1, 2016-05-05:
Even just including a "see also metadata.h for details" would be a big improvement. Duplicating the information in a bunch of places seems like it just leads to confusion.

jynik, 2016-05-28:
Tried to clarify this based upon points discussed here.  Committed in a83c62d92d9e49c1cffaaf970a034eb819d1a95f
and available in [libbladeRF v1.7.1 docs](https://nuand.com/libbladeRF-doc/v1.7.1/group___f_m_t___m_e_t_a.html#gga4c61587834fd4de51a8e2d34e14a73b2a4af3974535ba338bbcecaa8d7500de03).

If this is still terribly confusing, let me know.  

I'm a bit torn WRT information duplication - I would like the API docs to be relatively comprehensive on their own.

jynik, 2016-05-28:
Sigh. I see there's trailing `]` in the 1.7.1 docs. Fixing this now.

closed 2016-05-28 by jynik