Nuand/bladeRF #473: [Windows][libbladeRF] Triggered TX Stalls on Cypress Backend

jynik, 2016-05-22:
I've observed that when using the triggering functionality introduced in libbladeRF v1.7.0 with the Cypress driver backend, the library hangs when a trigger is issued.

The libusb backend in Windows does not exhibit this failure.  Those affected by the aforementioned Cypress driver hang can work around it by using the libusb drivers provided in the bladeRF windows installer, or provided via [Zadig](http://zadig.akeo.ie/).

Thus, to avoid further delaying this functionality for users not affected by this, libbladeRF v1.7.0 has been released and I will further investigate a fix for this for v1.7.1 release.

jynik, 2016-06-21:
I suspect this is effectively a duplicate of #443.

rtucker, 2017-08-02:
duplicate of #443

closed 2017-08-02 by rtucker