Nuand/bladeRF #475: [libbladeRF] Windows 10 builds fail if CyFX3 SDK is present

jynik, 2016-06-09:
To the best of my knowledge, the Cypress FX3 SDK 1.3.3 does not support Windows 10.  

`FindCyAPI.cmake` will report this a `FATAL_ERROR` when performing Windows builds if the FX3 SDK is installed on the system, even when `ENABLE_CYAPI_BACKEND=false`. 

This should be changed to a warning and let the "Was the CyAPI found?" logic take care of a fatal error, only if this backend was requested to be built.

rtucker, 2017-08-07:
Cypress does now support-ish:

http://www.cypress.com/knowledge-base-article/drivers-ez-usb-fx1-fx2lp-and-fx3-kba94413

rtucker, 2017-08-07:
so kinda got it, but can't work through the device signing requirements.

nonetheless, going to land most of the support for this

rtucker, 2017-08-07:
Pull req #522 resolves this issue, theoretically

closed 2017-08-07 by rtucker