Nuand/bladeRF #477: yatebts something error?

1024xroot, 2016-07-05:
in wiki :[https://github.com/Nuand/bladeRF/wiki/Minimalistic-build-and-run-test-for-OpenBTS-5#updates]

vim ./yatebts/mbts/TransceiverRAD1/bladeRFDevice.cpp

but directories of mbts can't find TransceiverRAD1

jynik, 2016-07-05:
The Yate team dropped support for that transceiver, and rewrote an entirely new transceiver.

The new transceiver is in their `yate` repo, not the `yatebts`.  (I believe this is so that the bladeRF transceiver implementation could be used for their LTE work.)

Therefore, this wiki page  documentation would need to be updated to point to the last usable `yatebts` SVN revision in which the bladeRFDevice.cpp is still present.

jynik, 2016-07-05:
I would recommend reviewing the dates of when the aforementioned wiki page was worked on, and use that to locate a sufficiently old YateBTS SVN revision.

1024xroot, 2016-07-05:
Thanks a lot

jynik, 2016-07-05:
@1024xroot Once you track down the SVN commit, could you share it here, or even better, update the wiki and let me know to close this?

mnhauke, 2016-07-05:
@jynik I'll update the wiki regarding this topic soon

mnhauke, 2016-07-12:
I've made a comment in the Wiki.
@1024xroot You need YateBTS SVN revision 503 or older! In later SVN revisions the RAD1 based transceiver that can be used with OpenBTS or OsmoBTS was removed.

jynik, 2016-07-12:
@mnhauke Thanks for tracking that commit down and adding it to the wiki.  Closing this issue down.

closed 2016-07-12 by jynik

1024xroot, 2016-07-13:
@mnhauke @jynik thanks I'll try again