Nuand/bladeRF #482: [bladeRF-cli] bladeRF-cli hangs on usb access on FreeBSD

kjopek, 2016-08-06:
On FreeBSD bladeRF-cli hangs while getting configuration of FPGA. LED1 & LED3 are on, LED2 is blinking.

Here is the backlog:

```
Program received signal SIGINT, Interrupt.
0x00000008013fcb1a in _poll () from /lib/libc.so.7
(gdb)  bt
#0  0x00000008013fcb1a in _poll () from /lib/libc.so.7
#1  0x0000000800a9fd66 in pthread_suspend_all_np () from /lib/libthr.so.3
#2  0x0000000801115709 in libusb_handle_events () from /usr/lib/libusb.so.3
#3  0x000000080111549b in libusb_handle_events_timeout_completed () from /usr/lib/libusb.so.3
#4  0x0000000801115c72 in libusb_bulk_transfer () from /usr/lib/libusb.so.3
#5  0x0000000801115b62 in libusb_bulk_transfer () from /usr/lib/libusb.so.3
#6  0x0000000800881fe4 in lusb_bulk_transfer (driver=0x802024040, endpoint=2 '\002', buffer=0x7fffffffe3b0, buffer_len=16, timeout_ms=250)
    at /tmp/bladeRF/host/libraries/libbladeRF/src/backend/usb/libusb.c:827
#7  0x000000080087af3e in nios_access (dev=0x80201d100, buf=0x7fffffffe3b0 "C\001") at /tmp/bladeRF/host/libraries/libbladeRF/src/backend/usb/nios_access.c:70
#8  0x000000080087b4ab in nios_8x32_read (dev=0x80201d100, id=1 '\001', addr=0 '\0', data=0x7fffffffe438)
    at /tmp/bladeRF/host/libraries/libbladeRF/src/backend/usb/nios_access.c:201
#9  0x000000080087b799 in nios_config_read (dev=0x80201d100, val=0x7fffffffe438) at /tmp/bladeRF/host/libraries/libbladeRF/src/backend/usb/nios_access.c:295
#10 0x000000080085cf00 in init_device (dev=0x80201d100) at /tmp/bladeRF/host/libraries/libbladeRF/src/bladerf_priv.c:144
#11 0x00000008008595a4 in bladerf_open_with_devinfo (opened_device=0x802016080, devinfo=0x7fffffffe540) at /tmp/bladeRF/host/libraries/libbladeRF/src/bladerf.c:234
#12 0x0000000800859b38 in bladerf_open (device=0x802016080, dev_id=0x0) at /tmp/bladeRF/host/libraries/libbladeRF/src/bladerf.c:282
#13 0x0000000000405621 in open_device (rc=0x7fffffffe650, state=0x802016080, status=0) at /tmp/bladeRF/host/utilities/bladeRF-cli/src/main.c:302
#14 0x000000000040539c in main (argc=3, argv=0x7fffffffe700) at /tmp/bladeRF/host/utilities/bladeRF-cli/src/main.c:479
```

jynik, 2016-08-10:
@kjopek, I believe you may want to chat with @darkfiberiru to compare notes.  IIRC, he had been seeing similar issues and was working on libusb patches.

darkfiberiru, 2016-08-13:
This is fixed in freebsd 11 thanks to HPS.(See freebsd commit details below) It is marked to MFC(backport)  to 10 and should go in cleanly but last time I checked it hasn't happened and I ASSUME that even if that happens it wouldn't be seen fixed until 10.4 though someone could probably apply the fix themselves if you have a particularly urgent need.

I am working on a port and further testing of bladerf on freebsd 10 since this got fixed. I had gotten side track in the last two months with personal stuff but am trying to put some more time in this weekend/week to finish up the port submission.

> Author: hselasky
> Date: Sun Jun 12 23:26:38 2016
> New Revision: 301842
> URL: https://svnweb.freebsd.org/changeset/base/301842
> 
> Log:
>   Implement code to stop all USB endpoints before executing a USB device
>   reset command, alternate setting command or set configuration
>   command. Else LibUSB v1.0 will not re-open the endpoints which the
>   kernel closes and the USB application might wait infinitely for
>   transfers to complete.
> 
>   Approved by:  re (hrs)
>   MFC after:    3 days
> 
> Modified:
>   head/lib/libusb/libusb10.c
> 
> Modified: head/lib/libusb/libusb10.c
> ==============================================================================
> --- head/lib/libusb/libusb10.c  Sun Jun 12 22:55:50 2016        (r301841)
> +++ head/lib/libusb/libusb10.c  Sun Jun 12 23:26:38 2016        (r301842)
> @@ -51,6 +51,8 @@
>  #include "libusb.h"
>  #include "libusb10.h"
> 
> +#define        LIBUSB_NUM_SW_ENDPOINTS (16 * 4)
> +
>  static pthread_mutex_t default_context_lock = PTHREAD_MUTEX_INITIALIZER;
>  struct libusb_context *usbi_default_context = NULL;
> 
> @@ -442,7 +444,7 @@ libusb_open(libusb_device *dev, libusb_d
>         if (dev == NULL)
>                 return (LIBUSB_ERROR_INVALID_PARAM);
> 
> -       err = libusb20_dev_open(pdev, 16 * 4 /* number of endpoints */ );
> +       err = libusb20_dev_open(pdev, LIBUSB_NUM_SW_ENDPOINTS);
>         if (err) {
>                 libusb_unref_device(dev);
>                 return (LIBUSB_ERROR_NO_MEM);
> @@ -1482,7 +1484,17 @@ libusb_cancel_transfer(struct libusb_tra
>  UNEXPORTED void
>  libusb10_cancel_all_transfer(libusb_device *dev)
>  {
> -       /* TODO */
> +       struct libusb20_device *pdev = dev->os_priv;
> +       unsigned x;
> +
> +       for (x = 0; x != LIBUSB_NUM_SW_ENDPOINTS; x++) {
> +               struct libusb20_transfer *xfer;
> +
> +               xfer = libusb20_tr_get_pointer(pdev, x);
> +               if (xfer == NULL)
> +                       continue;
> +               libusb20_tr_close(xfer);
> +       }
>  }
> 
>  uint16_t

kjopek, 2016-08-15:
Thank you very much for pointing out the solution. I simply created my own compilation of libusb and it seems to work correctly.

closed 2016-08-15 by kjopek