Nuand/bladeRF #483: LibUSB ep_to_pipeRef fails if using GHC

wyager, 2016-08-11:


wyager, 2016-08-11:
OK, so it looks like the "latest" download links aren't actually the latest. I've switched to FPGA firmware 4d3a2f8adb459b67391aa0a6225461df2fbb8e3fb2ae7589b9dd22c8c1b0041b and FX3 firmware 24d82869e658d4e9f4e57d87f42188db1a184804ad8e827e2302d45addb25a00. The problem persists. Failure in nios_legacy_get_fpga_version (via the code path mentioned above), but only when my Haskell program is compiled in profiling mode, which presumably could be doing something with e.g. stack layout.

bpadalino, 2016-08-11:
I am not sure what we could be doing but what type of failure is happening in `ep_to_pipeRef` to fail?  Is it a timeout of some type or an actual bulk transfer failure?

Unfortunately I know absolutely nothing about Haskell or `pipes-bladerf`.  Is there a maintainer you can hit with a question?

wyager, 2016-08-12:
`ep_to_pipeRef` is a fairly low-level LibUSB function. Looks like it converts a LibUSB endpoint to an OS X USB Pipe. Not sure what the significance of that is.

Unfortunately, those projects are even farther away from the USB stack than libbladeRF is, so if you don't know, they definitely won't have any idea. If nothing about the call stack I mentioned stands out to you, I'll probably move down the stack to the LibUSB folks.

jynik, 2016-08-12:
You may want to just extract the bare minimum code out of libbladeRF to reproduce it -- simply opening a device handle and issuing the single bulk transfer that's erroring out.

If this is only happening with the Haskell wrapper and -prof build, I'd be more prone to wonder if you're seeing memory or stack corruption, as you noted.  Getting yourself a debug build (or at least symbols) might help inspect differences with/without -prof.

closed 2017-08-02 by rtucker