Nuand/bladeRF #484: [libbladeRF] Windows 10 allows multiple device handles for single device (libusb)

ifrasch, 2016-08-11:
In past versions of Windows, only a single process could have an open handle to a device using libusb (this is the case in Linux). In Windows 10, multiple processes can now open a device handle to the same bladeRF device. This is bad because if you are using multiple devices and want to open them, in the past you could simply open "any available device" twice and it would open each device separately. Now it will open one of the devices twice. This issue can be reproduced using bladeRF-cli:

1) Plug one bladeRF device into a Windows 10 PC
2) Run "bladeRF-cli -i" on cmd.exe
3) Open another cmd.exe window, and run "bladeRF-cli -i"
4) Instead of throwing an error, the second window will run bladeRF-cli on the same device.

You can run even more instances of bladeRF-cli and it will still work. You can even set the frequency/gains in one window, and then when you type "print" in the other windows, the frequency/gain will show up.

ifrasch, 2016-08-11:
A similar issue was seen in the Cypress backend and it was resolved using a per device global mutex:
https://github.com/Nuand/bladeRF/blob/da239d19cc756b5ab7c2fc91ff3f3e86c3326833/host/libraries/libbladeRF/src/backend/usb/cyapi.c#L95

ifrasch, 2016-08-29:
Bad news. I tried this on Windows 7 Enterprise and saw the same issue. 

Compiled bladeRF code with Visual Studio 12; used libusb backend. Same issue is seen with the CLI and other libbladeRF programs.

rtucker, 2017-08-05:
This should be fixed now.  Thanks!!  -rt

PR #519 (2017-08-05): Enforce exclusive access to bladeRF when using libusb under Windows
Fixes #484
closed 2017-08-05 by rtucker