Nuand/bladeRF #489: Options for building different host components

darkfiberiru, 2016-09-14:
Hello,

I'm working on the freebsd port for bladerf now that the usb issues are all worked out. Since all the components are versioned separately I think it would be best to track them separately (if we don't then version numbers make no sense).  I don't currently see functionality to separate out the various host components. At the bottom of this is my implementation which shouldn't effect any current builds but will allow you to do something like the following which will allow you to only build the cli (assuming the library is already built and will be linked against).

```

CMAKE_ARGS+= -DENABLE_FULL_HOST_BUILD:BOOL=OFF
CMAKE_ARGS+= -DENABLE_BLADERF-CLI_BUILD:BOOL=ON

```

I will submit a complete pull request later if wanted but first I wanted to see if there was  any interest in this at all and if there was thoughts on better ways to do it or semantics on what I did that could be improved(naming of options potentially)

 

```
--- host/CMakeLists.txt.orig	2016-06-29 15:55:21 UTC
+++ host/CMakeLists.txt
@@ -39,6 +39,23 @@ set(BLADERF_HOST_COMMON_SOURCE_DIR ${CMA
 # CMake >= 2.8.4 no longer defines WIN32. Acknowledge this change.
 set(CMAKE_LEGACY_CYGWIN_WIN32 0)

+
+#####################################################################
+# Choose which host components to build
+#####################################################################
+
+# Overriding these options allows components to be built and packaged separately
+option(ENABLE_FULL_HOST_BUILD "Build all of bladeRF host components" ON)
+
+if(ENABLE_FULL_HOST_BUILD)
+    option(ENABLE_BLADERF-CLI_BUILD "Build the bladeRF-cli component" ON)
+    option(ENABLE_LIBBLADERF_BUILD "Build the libbladeRF component" ON)
+else(ENABLE_FULL_HOST_BUILD)
+    option(ENABLE_BLADERF-CLI_BUILD "Build the bladeRF-cli component" OFF)
+    option(ENABLE_LIBBLADERF_BUILD "Build the libbladeRF component" OFF)
+endif(ENABLE_FULL_HOST_BUILD)
+
+
 ###############################
 # Misc. options
 ###############################
--- host/libraries/CMakeLists.txt.orig	2016-06-29 15:55:21 UTC
+++ host/libraries/CMakeLists.txt
@@ -1,3 +1,8 @@
 cmake_minimum_required(VERSION 2.8)
-add_subdirectory(libbladeRF)
-add_subdirectory(libbladeRF_test)
+
+message( "Pre libbladerf")
+if( ENABLE_LIBBLADERF_BUILD )
+  add_subdirectory(libbladeRF)
+  message( "Added libbladerf")
+  add_subdirectory(libbladeRF_test)
+endif(ENABLE_LIBBLADERF_BUILD)
--- host/utilities/CMakeLists.txt.orig	2016-06-29 15:55:21 UTC
+++ host/utilities/CMakeLists.txt
@@ -1,3 +1,5 @@
 cmake_minimum_required(VERSION 2.8)

+if(ENABLE_BLADERF-CLI_BUILD)
 add_subdirectory(bladeRF-cli)
+endif(ENABLE_BLADERF-CLI_BUILD)
root@test11:/usr/local/poudriere/ports/default/comms/bladeRF-cli # head files/*
==> files/patch-host_CMakeLists.txt <==
--- host/CMakeLists.txt.orig	2016-06-29 15:55:21 UTC
+++ host/CMakeLists.txt
@@ -39,6 +39,23 @@ set(BLADERF_HOST_COMMON_SOURCE_DIR ${CMA
 # CMake >= 2.8.4 no longer defines WIN32. Acknowledge this change.
 set(CMAKE_LEGACY_CYGWIN_WIN32 0)

+
+#####################################################################
+# Choose which host components to build
+#####################################################################

==> files/patch-host_libraries_CMakeLists.txt <==
--- host/libraries/CMakeLists.txt.orig	2016-06-29 15:55:21 UTC
+++ host/libraries/CMakeLists.txt
@@ -1,3 +1,8 @@
 cmake_minimum_required(VERSION 2.8)
-add_subdirectory(libbladeRF)
-add_subdirectory(libbladeRF_test)
+
+message( "Pre libbladerf")
+if( ENABLE_LIBBLADERF_BUILD )
+  add_subdirectory(libbladeRF)

==> files/patch-host_utilities_CMakeLists.txt <==
--- host/utilities/CMakeLists.txt.orig	2016-06-29 15:55:21 UTC
+++ host/utilities/CMakeLists.txt
@@ -1,3 +1,5 @@
 cmake_minimum_required(VERSION 2.8)

+if(ENABLE_BLADERF-CLI_BUILD)
 add_subdirectory(bladeRF-cli)
+endif(ENABLE_BLADERF-CLI_BUILD)
```

darkfiberiru, 2017-08-07:
I wanted to check if there is interest in this addition I'm still open to doing the work but need to know if it's something the project is even interested in/ has any comments on.  @rtucker are you the one I should be talking to now about this?

Thanks,

Nick 'darkfiberiru' Wolff

darkfiberiru, 2017-08-07:
This work should also cover issue #418  (Add CMake project config files) as I believe they will be needed when compiling components separately.

This will also lead to a freebsd port(pkg) with me as the maintainer. I previously submitted build patches for freebsd and worked with the freebsd usb guru to fix some libusb issues. Only thing I've been waiting for is to package bladerf neatly as is expected in freebsd ports tree.

rtucker, 2017-08-30:
Greetings!

So, there is some development underway for a major version bump of most of the host components, but it looks like this should work out OK.  It would also be great to get into ports!

(Apologies for the really slow responses to this... been working on new stuff and not preening the issues queue as much as I ought :frowning_face:)

darkfiberiru, 2018-08-10:
Oops I missed that this was responded to. Will touch base with someone here at defcon make sure what I'm thinking is in line with what makes sense

darkfiberiru, 2018-08-12:
First step of breaking out components into sane groups. (libbladerf bladerf-cli bladerf-fsk bindings etc...) https://github.com/darkfiberiru/bladeRF/commit/43fd331de1bf702a27647889b0fc9b1bfef43503