Nuand/bladeRF #493: Build errors on macOS 10.12 Sierra: clockid_t

Moisan, 2016-11-05:
macOS 10.12 defines `clockid_t`. Trying to build libbladerf I get the following error:

```
In file included from /tmp/libbladerf-20161105-29826-m5ighn/bladeRF-2016.06/host/libraries/libbladeRF/src/dc_cal_table.c:52:
In file included from /tmp/libbladerf-20161105-29826-m5ighn/bladeRF-2016.06/host/libraries/libbladeRF/src/dc_cal_table.h:26:
In file included from /tmp/libbladerf-20161105-29826-m5ighn/bladeRF-2016.06/host/libraries/libbladeRF/src/bladerf_priv.h:39:
/tmp/libbladerf-20161105-29826-m5ighn/bladeRF-2016.06/host/common/include/osx/clock_gettime.h:38:13: error: typedef redefinition with different types ('int' vs 'enum clockid_t')
typedef int clockid_t;
            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/time.h:171:3: note: previous definition is here
} clockid_t;
  ^
1 error generated.
make[2]: *** [libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/dc_cal_table.c.o] Error 1
make[1]: *** [libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  8%] Built target libbladeRF_test_fw_check
make: *** [all] Error 2
```

marcelokk, 2016-12-04:
Hello,

I had the same problem as you. To be able to build I commented some lines at clock_gettime.h.

The path to the file is bladeRF-master/host/common/include/osx/clock_gettime.h.

And what I commented is:
```
 /*
typedef int clockid_t;
#define CLOCK_REALTIME 0

int clock_gettime(clockid_t clk_id, struct timespec *tp);
*/
```

But I have just started working with this code so I am not sure about the side effects.

rtucker, 2017-08-02:
Greetings...

So I believe this was fixed in master as of 21690e5, but if y'all are still around, I've got a more general fix in the works with commit 0779109841a560316e42a49d5723a8f66a1d89f4 on the rtucker-issue-508 branch.  I'd like to make sure this still builds properly on a few different macOS versions.

If all goes well, during the "cmake .." step, you should see `-- Looking for clock_gettime in c - found` when building on 10.12 or later, and `-- Looking for clock_gettime in c - not found` and a message about `Including clock_gettime.c shim for macOS` on earlier versions.

Thanks!  -rt

rtucker, 2017-08-03:
(update: 236753aaea9278278c89a285d861ae8271d615d7 is the HEAD of the rtucker-issue-508 branch)

rtucker, 2017-08-04:
This should be fixed in master as of pull request #513.  Thanks for the report!  -rt

closed 2017-08-04 by rtucker