Nuand/bladeRF #496: Modelsim simulation

gustad, 2017-01-21:
Is the compile.do script up to date?

I get some errors if I try to run the simulation:

```
$ git rev-parse --short HEAD
fcd775f
$ cd ./hdl/fpga/platforms/bladerf/modelsim/
$ vsim -do compile.do < /dev/null
Reading pref.tcl
....
# <EOF>
$ grep Error: transcript 
# ** Error: (vcom-11) Could not find work.reset_synchronizer.
# ** Error: ../vhdl/bladerf-hosted.vhd(413): Unknown expanded name.
# ** Error: (vcom-11) Could not find work.reset_synchronizer.
# ** Error: ../vhdl/bladerf-hosted.vhd(423): Unknown expanded name.
# ** Error: (vcom-11) Could not find work.reset_synchronizer.
# ** Error: ../vhdl/bladerf-hosted.vhd(433): Unknown expanded name.
# ** Error: (vcom-11) Could not find work.reset_synchronizer.
# ** Error: ../vhdl/bladerf-hosted.vhd(443): Unknown expanded name.
# ** Error: (vcom-11) Could not find work.reset_synchronizer.
# ** Error: ../vhdl/bladerf-hosted.vhd(584): Unknown expanded name.
# ** Error: (vcom-11) Could not find work.trigger.
# ** Error: ../vhdl/bladerf-hosted.vhd(763): Unknown expanded name.
# ** Error: (vcom-11) Could not find work.reset_synchronizer.
# ** Error: ../vhdl/bladerf-hosted.vhd(782): Unknown expanded name.
# ** Error: (vcom-11) Could not find work.trigger.
# ** Error: ../vhdl/bladerf-hosted.vhd(792): Unknown expanded name.
# ** Error: (vcom-11) Could not find work.signal_generator.
# ** Error: ../vhdl/bladerf-hosted.vhd(840): Unknown expanded name.
# ** Error: ../vhdl/bladerf-hosted.vhd(1120): VHDL Compiler exiting
```

Or do I have to run a command to generate the unknown modules?

Nuand, 2017-01-23:
compile.do is up to date.
Could you please output the contents of vsim to a file and upload it?

gustad, 2017-01-23:
https://gist.github.com/gustad/3cfe9bff3bc09470046d3fa2fd637212

```
git clone https://github.com/Nuand/bladeRF.git
cd bladeRF/hdl/fpga/platforms/bladerf/modelsim
vsim -do compile.do
```

bglod, 2017-02-20:
Thanks for reporting this. FIxed in 16c9e0966e16fb52c33b20ef9679e68c10fba84e.

closed 2017-02-20 by bglod