Nuand/bladeRF #499: [hdl] IQ phase/gain corrections not synchronized to proper clock domains

ifrasch, 2017-03-23:
The signals _correction_tx_phase_, _correction_tx_gain_, _correction_rx_phase_, and _correction_rx_gain_ in the top level bladerf-hosted.vhd architecture should be synchronized with the TX/RX sampling clocks before being fed into the rx_iq_correction and tx_iq_correction modules. Right now they are not synchronized and are coming directly from the 80 MHz Nios clock.

If there's something I'm missing let me know, but this looks like it could potentially cause metastability problems and corrupt a few samples if the user is applying IQ gain/phase correction.

bglod, 2017-04-19:
You are correct! A handshake block needs to be added someplace, either inside the iq_correction.vhd or in the top-level. Probably at the top-level because the iq_correction.vhd assumes everything is in its clock domain already.

rtucker, 2018-09-12:
Verified that this is still an unsynchronized clock domain crossing in the adsb, tx, and hosted revisions for the bladeRF (non-micro) w/ latest master.  Also looks like it's set_false_path'd in bladerf.sdc.