Nuand/bladeRF #502: Problem while upgrading firmware using the FX3 bootloader

darkbyte-ru, 2017-05-19:
When I run bladeRF-cli, I got message about firmware upgrade needed, but I have a problem with that.
```
c:\Program\bladeRF\x64>bladeRF-cli.exe -i
[INFO @ usb.c:367] FX3 FW v1.5.3 does not support the "device ready" query.
        Ensure flash-autoloading completes before opening a device.
        Upgrade the FX3 firmware to avoid this message in the future.

[WARNING @ bladerf.c:182] Firmware v1.5.3 was detected. libbladeRF v1.7.2 requires firmware v1.6.0 or later. An upgrade via the
bootloader is required.

bladeRF>
```

I follow steps from [wiki](https://github.com/Nuand/bladeRF/wiki/Upgrading-bladeRF-firmware):

1) Unplug bladerf
2) Connect jumper J64
3) Plug bladerf
4) Replace driver to libusb using Zadig
5) Remove jumper J64
6) Run bladeRF-cli.exe -i
```
c:\Program\bladeRF\x64>bladeRF-cli.exe -i
NOTE: One or more FX3-based devices operating in bootloader mode
      were detected. Run 'help recover' to view information about
      downloading firmware to the device(s).

No bladeRF device(s) available.

If one is attached, ensure it is not in use by another program
and that the current user has permission to access it.

bladeRF> recover

  FX3 bootloader devices:
  ---------------------------------------------------------
    Backend:    libusb
    Bus:        2
    Address:    8

  Use 'recover <bus> <addr> <firmware>' to download and boot
  firmware to the specified device.

bladeRF> recover 2 8 c:\Program\bladeRF\fx3_images\latest.img

  Success! Use "open" to switch to this device.
  Note that a "load fx3 <firmware>" is required to write the firmware to flash.

bladeRF> open

  Error: No devices available
```

On board light only one LED, and three status leds are off.

Again repeat steps 1-6, but skip the 4:
```
bladeRF> recover

  FX3 bootloader devices:
  ---------------------------------------------------------
    Backend:    Cypress driver
    Bus:        0
    Address:    1

  Use 'recover <bus> <addr> <firmware>' to download and boot
  firmware to the specified device.

bladeRF> recover 0 1 c:\Program\bladeRF\fx3_images\latest.img

  Success! Use "open" to switch to this device.
  Note that a "load fx3 <firmware>" is required to write the firmware to flash.

bladeRF> open

  Error: File or device I/O failure
```
After "recover" command three status leds is switchs on, and whine "open" command the middle led is blink.

Instead of latest.imgI also try bladeRF_fw_v1.9.1.img and bladeRF_fw_v2.0.0.img, and result the same.

What am I doing wrong?

bglod, 2017-05-19:
I'm not sure; I can't say that I've seen this error during the recovery process.

When debugging issues like this, run the CLI in either `debug` or `verbose` output mode:

`bladeRF-cli.exe -v debug -i` or `bladeRF-cli.exe -v verbose -i`

Some other things to check:
- Did you reboot after changing the driver?
    - The Wiki doesn't explicitly say to, but Windows can be weird like that.
- Try another port on a **different** USB controller (USB 2.0 or USB 3.0, but USB 3.0 is recommended)
    - Performance and power delivery can vary depending on the controller. Some controllers are known to be [problematic](https://github.com/Nuand/bladeRF/wiki/Troubleshooting#problematic-configurations) with the bladeRF.
- Yours is probably fine, but try another USB cable if you happen to have one handy

I know other customers have been able to recover in Windows, so this should work. My only other thought is to try Linux since the drivers and whatnot are more robust in that environment. The GNU Radio Live Image is a bootable Ubuntu image with built-in support for bladeRF, so you can just boot it and run the CLI to update the firmware without having to modify your disk(s) or install/configure anything.

darkbyte-ru, 2017-05-19:
Ok, output with "-v debug" flag:
```
c:\Program\bladeRF\x64>bladeRF-cli.exe -v debug -i
[DEBUG @ libusb.c:292] Could not open device: LIBUSB_ERROR_NOT_SUPPORTED
NOTE: One or more FX3-based devices operating in bootloader mode
      were detected. Run 'help recover' to view information about
      downloading firmware to the device(s).

[DEBUG @ libusb.c:550] No devices available on the libusb backend.

No bladeRF device(s) available.

If one is attached, ensure it is not in use by another program
and that the current user has permission to access it.

bladeRF> recover
[DEBUG @ libusb.c:292] Could not open device: LIBUSB_ERROR_NOT_SUPPORTED

  FX3 bootloader devices:
  ---------------------------------------------------------
    Backend:    Cypress driver
    Bus:        0
    Address:    1

  Use 'recover <bus> <addr> <firmware>' to download and boot
  firmware to the specified device.
```
Currently defalt cypress driver installed, according to "recover" output, so use Zadig and change it to libusb, after that output of "recover" command changes:

```
bladeRF> recover

  FX3 bootloader devices:
  ---------------------------------------------------------
    Backend:    libusb
    Bus:        2
    Address:    17

  Use 'recover <bus> <addr> <firmware>' to download and boot
  firmware to the specified device.

bladeRF> recover 2 17 c:\Program\bladeRF\fx3_images\latest.img

  Success! Use "open" to switch to this device.
  Note that a "load fx3 <firmware>" is required to write the firmware to flash.
```

After writing new firmware to ram, device ID changes and windows again install cypress driver. Try to "open":
```
bladeRF> open
[DEBUG @ libusb.c:436] Could not open bladeRF device: LIBUSB_ERROR_NOT_SUPPORTED
[DEBUG @ libusb.c:550] No devices available on the libusb backend.
[DEBUG @ cyapi.c:409] Transfered failed.
[DEBUG @ nios_legacy_access.c:123] Failed to receive NIOS II response: File or device I/O failure
[DEBUG @ nios_legacy_access.c:213] Failed to read FPGA version[0]: File or device I/O failure

  Error: File or device I/O failure
```

Now change driver to libusb and again "open":
```
bladeRF> open
[DEBUG @ nios_legacy_access.c:123] Failed to receive NIOS II response: Operation timed out
[DEBUG @ nios_legacy_access.c:213] Failed to read FPGA version[0]: Operation timed out

  Error: Operation timed out
```

I try another USB 3.0 port on my laptop, also try different cable, without any success.

Also same output with verbose flag:
```
bladeRF> recover
[VERBOSE @ libusb.c:250] Found an FX3 bootloader.
[VERBOSE @ libusb.c:310] Added instance 0 to device list

  FX3 bootloader devices:
  ---------------------------------------------------------
    Backend:    libusb
    Bus:        2
    Address:    22

  Use 'recover <bus> <addr> <firmware>' to download and boot
  firmware to the specified device.

bladeRF> recover 2 22 c:\Program\bladeRF\fx3_images\latest.img
[VERBOSE @ fx3_fw.c:196] Scanned section 0 at offset 0x00000004: addr=0x00000100, len=0x0000088a
[VERBOSE @ fx3_fw.c:196] Scanned section 1 at offset 0x00002234: addr=0x40003000, len=0x00004000
[VERBOSE @ fx3_fw.c:196] Scanned section 2 at offset 0x0001223c: addr=0x40013000, len=0x00001f3e
[VERBOSE @ fx3_fw.c:196] Scanned section 3 at offset 0x00019f3c: addr=0x40030000, len=0x00000c18
[VERBOSE @ fx3_fw.c:214] Firmware checksum OK.
[VERBOSE @ libusb.c:688] Opened bootloader at 2:22
[VERBOSE @ usb.c:919] Writing 4096 bytes to bootloader @ 0x00000100
[VERBOSE @ usb.c:936] Reading back 4096 bytes from bootloader @ 0x00000100
[VERBOSE @ usb.c:919] Writing 4096 bytes to bootloader @ 0x00001100
[VERBOSE @ usb.c:936] Reading back 4096 bytes from bootloader @ 0x00001100
[VERBOSE @ usb.c:919] Writing 552 bytes to bootloader @ 0x00002100
[VERBOSE @ usb.c:936] Reading back 552 bytes from bootloader @ 0x00002100
....
[VERBOSE @ usb.c:936] Reading back 4096 bytes from bootloader @ 0x40031000
[VERBOSE @ usb.c:919] Writing 4096 bytes to bootloader @ 0x40032000
[VERBOSE @ usb.c:936] Reading back 4096 bytes from bootloader @ 0x40032000
[VERBOSE @ usb.c:919] Writing 96 bytes to bootloader @ 0x40033000
[VERBOSE @ usb.c:936] Reading back 96 bytes from bootloader @ 0x40033000
[VERBOSE @ usb.c:986] Booting new FW.

  Success! Use "open" to switch to this device.
  Note that a "load fx3 <firmware>" is required to write the firmware to flash.

bladeRF> open
[VERBOSE @ libusb.c:541] Using libusb version: 1.0.20.11004
[VERBOSE @ libusb.c:422] Found a bladeRF (idx=5)
[DEBUG @ libusb.c:436] Could not open bladeRF device: LIBUSB_ERROR_NOT_SUPPORTED
[DEBUG @ libusb.c:550] No devices available on the libusb backend.
[VERBOSE @ cyapi.c:189] Added instance 0 to device list
[VERBOSE @ cyapi.c:112] Mutex name: Global\bladeRF-21f0e6807765fd148ec86b58d0e65e43
[VERBOSE @ capabilities.c:45] Capability mask before FPGA load: 0x0000000700000000
[VERBOSE @ usb.c:112] Changing to USB alt setting 0
[VERBOSE @ usb.c:112] Changing to USB alt setting 1
[DEBUG @ cyapi.c:409] Transfered failed.
[DEBUG @ nios_legacy_access.c:123] Failed to receive NIOS II response: File or device I/O failure
[DEBUG @ nios_legacy_access.c:213] Failed to read FPGA version[0]: File or device I/O failure

  Error: File or device I/O failure

bladeRF> open
[VERBOSE @ libusb.c:541] Using libusb version: 1.0.20.11004
[VERBOSE @ libusb.c:422] Found a bladeRF (idx=5)
[VERBOSE @ capabilities.c:45] Capability mask before FPGA load: 0x0000000700000000
[VERBOSE @ usb.c:112] Changing to USB alt setting 0
[VERBOSE @ usb.c:112] Changing to USB alt setting 1
[DEBUG @ nios_legacy_access.c:123] Failed to receive NIOS II response: Operation timed out
[DEBUG @ nios_legacy_access.c:213] Failed to read FPGA version[0]: Operation timed out

  Error: Operation timed out
```

Ok, I will try flush from live linux image.

rtucker, 2017-08-07:
I was unable to reproduce this on Windows 10 Enterprise Evaluation (version 1703, build 15063.483, running in KVM (QEMU 2.8.0, kernel 4.10.0-28-generic)) with bladeRF-cli/libbladeRF at commit daab2034, libusb version 1.0.21.11156, and no CyAPI stuff installed.

I will install the Cypress SDK and see how things go there...

rtucker, 2017-08-07:
Reproduction currently blocking on #523

AnriSank, 2018-06-24:
I have same problem
[image]

rtucker, 2018-09-12:
Windows installers built after today (so, any release _after_ the 2018.08 release) will associate with the bootloader VID/PID during installation, allowing the recovery method to work with LibUSB, without manually using Zadig to set up the association.  The experience will (hopefully) match that on non-Windows platforms.

Thanks!  -rt

closed 2018-09-12 by rtucker

marwon34, 2019-01-25:
No bladeRF device(s) available.

If one is attached, ensure it is not in use by another program
and that the current user has permission to access it.

bladeRF>