Nuand/bladeRF #508: Problems during installing on Mac OS X 10.11.3

Veryjius, 2017-07-27:
Hello,I'm trying to install bladerf on my Mac running 10.11.3    I am running through the instructions in the [Wiki,](https://github.com/Nuand/bladeRF/wiki/Getting-started:-Mac-OSX), but I keep getting an error on the make step:
`
Scanning dependencies of target libbladerf_shared
[  1%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/bladerf_priv.c.o
/Users/mac/bladeRF/host/libraries/libbladeRF/src/bladerf_priv.c:291:14: error:
      implicit declaration of function 'clock_gettime' is invalid in C99
      [-Werror,-Wimplicit-function-declaration]
    
status = clock_gettime(CLOCK_REALTIME, t);
              ^
/Users/mac/bladeRF/host/libraries/libbladeRF/src/bladerf_priv.c:291:28: error:
      use of undeclared identifier 'CLOCK_REALTIME'
    status = clock_gettime(CLOCK_REALTIME, t);
                                         ^
2 errors generated.
make[2]: *** [libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/bladerf_priv.c.o] Error 1
make[1]: *** [libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/all] Error 2
make: *** [all] Error 2`

The ` ^` are under 'c' of clock_gettime and 'C' of CLOCK_REALTIME

I noticed that there were many similar methods to solve it on Google but it doesn't work on my Mac (like [https://nuand.com/forums/viewtopic.php?f=4&t=3494](url) which tried to changed clock_gettime  to clock_get_time. Please tell me how can I make this work?

txjacob, 2017-08-01:
I had a lot of problems getting everything working on my Mac when I was setting it up. I can't remember if I ever got it to work correctly when compiling for source, but if you don't need to compile it manually I strongly recommend installing through macports. It simplified the install process immensely for me.

Veryjius, 2017-08-01:
@txjacob Yes I tried macports before. But when I used `sudo port install bladeRF +tecla
` ,the terminal showed these:
`Warning: Xcodebuild exists but fial to execute`
`Warning:Xcode does not appear to be installed;most ports will likely fail to build.`
`---> Computing dependencies for bladeRF`
`---> Cleaning bladeRF`
`---> Scaning binaries for linking error`
`---> No broken files found.`

Dose it mean my Xcode doesn't install complety and leads this error? Can I reinstall Xcode and macports to try installing bladeRF through macports ?

rtucker, 2017-08-01:
Hmm... so it sounds like, for whatever reason, the wrapper around clock_gettime() is not being included.

Could you attach the output of "cmake -L" in your host/build/ directory, along with the output of "cpp -dM /dev/null" ?  I'm suspecting one of the indicators used to determine that you're building on macOS has changed.

rtucker, 2017-08-01:
The output of "cmake --system-information" is probably also interesting!

txjacob, 2017-08-02:
Have you installed the graphical version of xcode or the command line version?

Veryjius, 2017-08-02:
@txjacob I don't know. I installed Xcode in the store and the network was disconected a while.

@rtucker OK, I will try it later :)

Veryjius, 2017-08-02:
@rtucker  Here are the result.

`cmake -L` : 
`CMake Error: The source directory "/Users/mac/bladeRF/host/build" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
-- Cache values
BLADERF-FSK_BYPASS_PHY_SCRAMBLING:BOOL=OFF
BLADERF-FSK_BYPASS_RX_CHANNEL_FILTER:BOOL=OFF
BLADERF-FSK_BYPASS_RX_PNORM:BOOL=OFF
BLADERF-FSK_ENABLE_DEBUG_ALL:BOOL=OFF
BLADERF-FSK_ENABLE_DEBUG_BLADERF:BOOL=OFF
BLADERF-FSK_ENABLE_DEBUG_FSK:BOOL=OFF
BLADERF-FSK_ENABLE_DEBUG_LINK:BOOL=OFF
BLADERF-FSK_ENABLE_DEBUG_PHY:BOOL=OFF
BLADERF-FSK_ENABLE_DEBUG_TEST_SUITE:BOOL=OFF
BLADERF-FSK_ENABLE_NOTES_LINK:BOOL=OFF
BLADERF-FSK_ENABLE_NOTES_PHY:BOOL=OFF
BLADERF_GROUP:STRING=plugdev
BUILD_BLADERF_CLI_DOCUMENTATION:BOOL=OFF
BUILD_DOCUMENTATION:BOOL=OFF
BUILD_LIBBLADERF_DOCUMENTATION:BOOL=OFF
BUILD_LIBBLADERF_DOC_EXAMPLES:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=Release
CMAKE_INSTALL_PREFIX:PATH=/opt/local
CMAKE_OSX_ARCHITECTURES:STRING=
CMAKE_OSX_DEPLOYMENT_TARGET:STRING=
CMAKE_OSX_SYSROOT:STRING=
ENABLE_BACKEND_CYAPI:BOOL=OFF
ENABLE_BACKEND_DUMMY:BOOL=OFF
ENABLE_BACKEND_LIBUSB:BOOL=ON
ENABLE_BACKEND_USB:BOOL=ON
ENABLE_GDB_EXTENSIONS:BOOL=OFF
ENABLE_LIBBLADERF_LOGGING:BOOL=ON
ENABLE_LIBBLADERF_SYNC_LOG_VERBOSE:BOOL=OFF
ENABLE_LIBBLADERF_SYSLOG:BOOL=OFF
ENABLE_LIBTECLA:BOOL=ON
ENABLE_LOCK_CHECKS:BOOL=OFF
ENABLE_LOG_FILE_INFO:BOOL=ON
ENABLE_USB_DEV_RESET_ON_OPEN:BOOL=OFF
INSTALL_UDEV_RULES:BOOL=OFF
PANDOC_EXECUTABLE:FILEPATH=/usr/local/bin/pandoc
TREAT_WARNINGS_AS_ERRORS:BOOL=ON
UDEV_RULES_PATH:STRING=/etc/udev/rules.d`

`cpp -dM /dev/null` : 
`clang: warning: argument unused during compilation: '-dM/dev/null'`

`cmake --system-information` (It's sooooo long that I upload a txt ) : 
[systeminformation.txt](https://github.com/Nuand/bladeRF/files/1192744/systeminformation.txt)

rtucker, 2017-08-02:
looks like `CMAKE_SYSTEM_NAME == "Darwin"` in the system information, so there goes my first guess...

So that should set `BLAEDRF_OS_OSX` and then add `${BLADERF_HOST_COMMON_SOURCE_DIR}/osx/clock_gettime.c` to the build... it should also `#define BLADERF_OS_OSX 1` in `host/build/host_config.h`.

Next thing that has to happen is this condition has to be true in `host/common/include/osx/clock_gettime.h` and `host/common/src/osx/clock_gettime.c`:

`#if MAC_OS_X_VERSION_MAX_ALLOWED < MAC_OS_X_VERSION_10_12`

I don't know what those two variables are defined as in macOS 10.11.3, but if that condition isn't met, you're gonna have a bad time.  Quick test for this would be to replace that line in both files with `#if 1`.

If this works, could you scrounge around your system includes (probably `/usr/include` but not necessarily) for `AvailabilityMacros.h` and attach it?  I suspect the MAC_OS_X_VERSION_* macros may be to blame.

I wonder if there's a way to explicitly test for clock_gettime from the cmake side of the world...

rtucker, 2017-08-02:
My current wager: you're running Xcode 8.1 or later (based on "Apple LLVM version 8.0.0 (clang-800.0.42.1)"), which has support for macOS 10.12, and so `MAC_OS_X_VERSION_MAX_ALLOWED` will therefore default to `MAC_OS_X_VERSION_10_12`.  Wheeee...

Hoping to have a possible fix shortly.

rtucker, 2017-08-02:
Give commit 0779109841a560316e42a49d5723a8f66a1d89f4 a try?  You'll probably want to blow away your build/ dir just to be sure cmake gets the hint.

Thanks!  -rt

Veryjius, 2017-08-03:
@rtucker I tried commit 0779109 and it worked ! The bladeRF is detected on my mac now.
```
$   bladeRF-cli -p
    Backend:        libusb
    Serial:         87c48eacc703138f6daaa7331a54bedd
    USB Bus:        20
    USB Address:    9
```

And the board also work in gnuradio project now.
Thanks a lot :)

rtucker, 2017-08-03:
Awesome!!!

Just for completeness -- if you could try a clean build of 236753aaea9278278c89a285d861ae8271d615d7, it would be helpful... I forgot a couple minor details and broke 10.12+ builds, heh.  (Just "mkdir build && cd build && cmake .. && make" will be enough, no need to "make install")

Thanks!!!!

Veryjius, 2017-08-04:
@rtucker OK I made it ;) 
Thanks.

rtucker, 2017-08-04:
It is now in master, and I'll close this issue.  =^.^=

Thanks and have a great weekend!  -rt

closed 2017-08-04 by rtucker