Nuand/bladeRF #527: xb 300 pwr doesn't allowed while transmitting

Gigithecode, 2017-09-11:
On a bladRF 115K + xb 300
Command line 
>xb 300 pwr
doesn't work while transmitting ("device busy")
otherwise the resullting wording in NOT appropriate and really misleading... 
"PA output power: " shouldn't be better something like "PA Detector Voltage (dBmV):" ?
or, better, an output based from an extension of the following table from SE2605L_203009E specs 
VDET
P OUT = +30 dBm 0.90 1.00 1.15 V
P OUT = +26 dBm 0.65 0.73 0.82 V
P OUT = 5 dB 0.30 0.33 0.36 V

Reference topic:
[https://nuand.com/forums/viewtopic.php?f=4&t=3946&p=7383#p7383](url)

bglod, 2017-09-11:
@Gigithecode I clarified the title, I hope you don't mind.

To summarize what we need to do:
1. Determine why we can't pass command/control to the XB-300 during transmission. (I think I know why, but can we fix it?)
2. The minimum voltage (0.3 V) corresponds to approximately +5 dBm output power. This is misleading when nothing is being transmitted. Can we add a correction factor, or use a threshold below which we simply report "zero" output power?

Gigithecode, 2017-09-12:
No problem!