Nuand/bladeRF #529: bladeRF-win-installer-latest.exe does not run under Windows 10

rtucker, 2017-09-25:
Attempting to run the Windows installer under Windows 10 results in a modal error:

"This app can't run on your PC
To find a version for your PC, check with the software publisher."

After some digging, the actual reason is in the Application Log via Event Viewer:

```
Log Name:      Application
Source:        Wow64 Emulation Layer
Date:          9/25/2017 7:20:49 AM
Event ID:      1109
Task Category: None
Level:         Information
Keywords:      Classic
User:          N/A
Computer:      DESKTOP-34HGPQF
Description:
The program or feature "\??\C:\Users\Rey Tucker\Downloads\temp\bladeRF-win-installer-latest.exe" cannot start or run due to incompatibity with 64-bit versions of Windows. Please contact the software vendor to ask if a 64-bit Windows compatible version is available.
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Wow64 Emulation Layer" />
    <EventID Qualifiers="16384">1109</EventID>
    <Level>4</Level>
    <Task>0</Task>
    <Keywords>0x80000000000000</Keywords>
    <TimeCreated SystemTime="2017-09-25T14:20:49.511378100Z" />
    <EventRecordID>2127</EventRecordID>
    <Channel>Application</Channel>
    <Computer>DESKTOP-34HGPQF</Computer>
    <Security />
  </System>
  <EventData>
    <Data>\??\C:\Users\Rey Tucker\Downloads\temp\bladeRF-win-installer-latest.exe</Data>
  </EventData>
</Event>
```

Original report: https://nuand.com/forums/viewtopic.php?f=4&t=3965

Issue was reproduced on Windows 10 Enterprise Evaluation, build 15063.rs2_release.170317-1834, 64-bit, QEMU, Intel Core Processor (Haswell, no TSX)

JamesHagerman, 2017-10-19:
What needs to happen to update this build so it runs correctly under Windows 10?

zenrandom, 2017-10-21:
I don't know for certain, but my guess is that it starts with compiling a 64 bit version of the installer.  I just ran into this problem.  It would be fantastic if they published a new working installer.

txjacob, 2017-10-29:
Has anyone tried compiling it under windows 10 yet? I could try giving it a go if it would help.

rtucker, 2017-10-29:
Greetings...

Yep, master does build under Windows 10, with some quirks (in particular, pthread-win32 needs modified and rebuilt).  https://github.com/Nuand/bladeRF/wiki/Getting-Started:-Windows has been updated with Visual Studio 2017 instructions.

A new installer build is in the works (really!), but in the meantime, could you try the 2016.05-rc2-alpha2 installer?  I believe it should be functionally equivalent to 2016.06 and it seems to actually work on Windows 10:
https://www.nuand.com/windows_installers/bladeRF-win-installer-2016.05-rc2-alpha2.exe

Thanks!  -rt

txjacob, 2017-10-30:
So I actually tried the 2016.06 installer on Windows 10 Pro and had no problems with it. Admittedly that was not on the creators update as I had not upgraded yet, but I upgraded yesterday so I will try again and post my results.

zenrandom, 2017-10-30:
I tested this on the creators update and it has installed without issue.

rtucker, 2017-12-08:
In the end, the issue turned out to be file corruption of that specific file, combined with a few other Windows 10-related quirks making builds difficult.  We have successfully built (and tested) signed Windows installers.

Until a new version is tagged and released, the 2016.05-rc2-alpha2 installer will work fine.

Thank you for your patience!

closed 2017-12-08 by rtucker