Nuand/bladeRF #533: Not building man page for bladeRF-cli.

fzy139, 2017-12-02:
Hi,
I'm using the Raspberry 3b(Raspbian lite) to build the bladeRF environment.It turnes out to be that the bladeRF-cli command was not found, And when I install the libbladeRF,it suggests that:

`cmake -DCMAKE_BUILD_TYPE=Release -DAKE_INSTALL_PREFIX=/usr/local -DINSTALL_UDEV_RULES=ON ../
-- Checking libusb version...
-- libusb version: 1.0.21

-- Checking libc version...
-- libc version: 2.24

--
-- libbladeRF version: 1.8.0-git-6116d090
-- nuand bladeRF udev rules will be installed to '/etc/udev/rules.d' upon runng 'make install'
--
-- Will use pandoc to generate interactive help
-- Checking for module 'libtecla'
--   No package 'libtecla' found
-- Not building man page for bladeRF-cli.  (BUILD_BLADERF_CLI_DOCUMENTATION isFF)
-- Configured to build bladeRF-cli version: 1.5.0-git-6116d090
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bladeRF/host/build
root@raspberrypi:/home/bladeRF/host/build# 
`
I've tried wget and apt-get to install the libtecla and it's  succeessful.

Really need a hand! 
Thanks!

rtucker, 2017-12-04:
That is an informational message, and it won't block the building of bladeRF-cli or the rest of the host software.

It looks like cmake is working properly there.  What is the output from "make" and "make install"?  That will help figure out where bladeRF-cli is going.

Also: for libtecla support (which is used in bladeRF-cli's interactive mode), you'll want to install the `libtecla-dev` package and re-run cmake.  It's not required, but it is convenient.

fzy139, 2017-12-15:
thx! @rtucker rtucker 
Still，I reinstalled  but the command bladeRF-cli is not found.
make and make install seems right.
`pi@raspberrypi:~/bladeRF-master/host/build $ make
Scanning dependencies of target libbladerf_shared
[  1%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/async.c.o
[  1%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/backend/backend.c.o
[  2%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/bladerf.c.o
[  2%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/bladerf_priv.c.o
[  3%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/capabilities.c.o
[  3%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/config.c.o
[  4%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/dc_cal_table.c.o
[  4%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/device_identifier.c.o
[  5%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/devinfo.c.o
[  5%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/file_ops.c.o
[  5%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/flash.c.o
[  6%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/flash_fields.c.o
[  6%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/fpga.c.o
[  7%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/fx3_fw.c.o
[  7%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/fx3_fw_log.c.o
[  8%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/gain.c.o
[  8%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/image.c.o
[  9%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/init_fini.c.o
[  9%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/si5338.c.o
[ 10%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/smb_clock.c.o
[ 10%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/sync.c.o
[ 11%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/sync_worker.c.o
[ 11%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/trigger.c.o
[ 12%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/tuning.c.o
[ 12%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/version_compat.c.o
[ 12%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/xb.c.o
[ 13%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/__/__/common/src/sha256.c.o
[ 13%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/__/__/common/src/conversions.c.o
[ 14%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/__/__/common/src/log.c.o
[ 14%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/home/pi/bladeRF-master/fpga_common/src/lms.c.o
[ 15%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/home/pi/bladeRF-master/fpga_common/src/band_select.c.o
[ 15%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/__/__/common/src/parse.c.o
[ 16%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/backend/usb/nios_access.c.o
[ 16%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/backend/usb/nios_legacy_access.c.o
[ 17%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/backend/usb/usb.c.o
[ 17%] Building C object libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/backend/usb/libusb.c.o
[ 18%] Linking C shared library ../../output/libbladeRF.so
[ 18%] Built target libbladerf_shared
[ 19%] Linking C executable ../../../output/libbladeRF_test_async
[ 19%] Built target libbladeRF_test_async
[ 19%] Linking C executable ../../../output/libbladeRF_test_bootloader_recovery
[ 20%] Built target libbladeRF_test_bootloader_recovery
[ 20%] Linking C executable ../../../output/libbladeRF_test_c
[ 21%] Built target libbladeRF_test_c
[ 22%] Linking CXX executable ../../../output/libbladeRF_test_cpp
[ 22%] Built target libbladeRF_test_cpp
[ 22%] Linking C executable ../../../output/libbladeRF_test_ctrl
[ 29%] Built target libbladeRF_test_ctrl
[ 29%] Linking C executable ../../../output/libbladeRF_test_freq_hop
[ 31%] Built target libbladeRF_test_freq_hop
Scanning dependencies of target libbladeRF_test_fw_check
[ 32%] Building C object libraries/libbladeRF_test/test_fw_check/CMakeFiles/libbladeRF_test_fw_check.dir/__/__/libbladeRF/src/fx3_fw.c.o
[ 32%] Building C object libraries/libbladeRF_test/test_fw_check/CMakeFiles/libbladeRF_test_fw_check.dir/__/__/libbladeRF/src/file_ops.c.o
[ 33%] Building C object libraries/libbladeRF_test/test_fw_check/CMakeFiles/libbladeRF_test_fw_check.dir/__/__/__/common/src/log.c.o
[ 33%] Linking C executable ../../../output/libbladeRF_test_fw_check
[ 33%] Built target libbladeRF_test_fw_check
[ 34%] Linking C executable ../../../output/libbladeRF_test_open
[ 35%] Built target libbladeRF_test_open
[ 35%] Linking C executable ../../../output/libbladeRF_test_peripheral_timing
[ 36%] Built target libbladeRF_test_peripheral_timing
[... cut ...]
cmake -DCMAKE_INSTALL_PREFIX=~/user/local -DENABLE_BACKEND_USB=ON -DENABLE_BACKEND_LIBUSB=ON  -LIBUSB_PATH=/lib/arm-linux-gnueabihf ../
ldconfig`

During the cmake in first time,this error happened:
No libbladeRF backends are enabled.So I installed the libusb-dev & add some options to cmake.
Finally,the bladeRF-cli is still not found.

Is it the problem of debian on Raspberrypi? Maybe I should use the ubuntu to bladeRF?
I'm  working on a project so probably bladeRF on Ubuntu is more perfect？

Thx a lot~

fzy139, 2017-12-15:
UPDATE：It's okay now. I just rebuild and reboot for several times and the bladeRF-cli is working now.Thx!!!

closed 2018-04-30 by rtucker