Nuand/bladeRF #534: Why I got 32'hCDCDCDCD like this using signaltap?

chendongbox, 2017-12-02:
I use bladerf_sync_tx function and send all zeros IQ data, but in signalTap, I got a lot "cdcdcdcd", what's these data? Is it a bug? 
Can I remove these "cdcdcdcd"?
[image]
[image]

bglod, 2017-12-04:
Can you provide a code snippet?
What is happening between the two SignalTap captures?

chendongbox, 2017-12-05:
I use fpga v0.6 and recompile it, there is the problem, now i use  fpga v0.5 and no problem. 
Manybe there are some timing problem ?
I will test more time to make sure there is no problem with v0.5.

bglod, 2017-12-05:
That's strange. I did a cursory test on Linux using bladeRF-cli and FPGA from master to transmit a CSV containing "0, 0" and was not able to reproduce this. The bladeRF-cli TX calls `bladerf_sync_tx()`, so this test _should_ exercise the same host code.

The only HDL files that appear relevant to this issue are `fx3_gpif.vhd` and `bladerf-hosted.vhd`. Upon reviewing the `git diff fpga_v0.5.0 origin/master` for these two files, there were no direct changes to the way the TX sample FIFO is written. There was a minor change in the way `tx_enable` and `rx_enable` are assigned in `fx3_gpif.vhd` that could be relevant. Because I could not reproduce the issue, I don't think it is.

I assume your SignalTap sample clock is set to `tx_fifo:U_tx_sample_fifo|wrclk` or `fx3_pll:U_fx3_pll|c0` (they're the same). Try adding all the `fx3_gpif:U_fx3_gpif|state.*` signals, as well as the `fx3_gpif_in` bus to the same SignalTap instance. If you see the 0xCD..CD data coming in over `fx3_gpif_in`, then it's very likely that there's an issue with the data buffer or another parameter you're providing to `bladerf_sync_tx()`.

Of course, report back either way so we can help get to the bottom of it if it is indeed a bug!

chendongbox, 2017-12-06:
Yes, the 0xCD..CD data coming in over fx3_gpif_in, I think the data come from the host, maybe corrections and calibration ? In any case, I will check the data buffer again.

bglod, 2017-12-08:
What if you use the bladeRF-cli as a sanity check?

chendongbox, 2017-12-16:
At last，after many test,  there is a bug in our software. libbladerf is ok! 
Thank you very much.

closed 2017-12-16 by chendongbox