Nuand/bladeRF #544: error: passing argument 1 of ‘str2bool’ from incompatible pointer type

baguzajja, 2018-02-07:
/root/src/bladeRF/host/libraries/libbladeRF/src/config.c:197:23: error: passing argument 1 of ‘str2bool’ from incompatible pointer type [-Werror=incompatible-pointer-types]
         ok = str2bool(opt.value, &val);
                       ^~~
In file included from /root/src/bladeRF/host/libraries/libbladeRF/src/bladerf_priv.h:46:0,
                 from /root/src/bladeRF/host/libraries/libbladeRF/src/config.c:22:
/root/src/bladeRF/host/common/include/conversions.h:327:5: note: expected ‘const char **’ but argument is of type ‘char *’
 int str2bool(const char **str, bool *val);
     ^~~~~~~~
/root/src/bladeRF/host/libraries/libbladeRF/src/config.c:197:34: error: passing argument 2 of ‘str2bool’ from incompatible pointer type [-Werror=incompatible-pointer-types]
         ok = str2bool(opt.value, &val);
                                  ^
In file included from /root/src/bladeRF/host/libraries/libbladeRF/src/bladerf_priv.h:46:0,
                 from /root/src/bladeRF/host/libraries/libbladeRF/src/config.c:22:
/root/src/bladeRF/host/common/include/conversions.h:327:5: note: expected ‘_Bool *’ but argument is of type ‘uint32_t * {aka unsigned int *}’
 int str2bool(const char **str, bool *val);
     ^~~~~~~~
cc1: all warnings being treated as errors
libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/build.make:182: recipe for target 'libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/config.c.o' failed
make[2]: *** [libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/src/config.c.o] Error 1
CMakeFiles/Makefile2:140: recipe for target 'libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/all' failed
make[1]: *** [libraries/libbladeRF/CMakeFiles/libbladerf_shared.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

when i try to install, i get error like this

any body can help

thanks

baguzajja, 2018-02-07:
when i try to fix this code on config.c
with this code 

} else if (!strcasecmp(opt.key, "agc")) {
        val = str2bool(opt.value, &ok);
        if (ok != 0)
            return BLADERF_ERR_INVAL;

        gain_mode = (val & 1) ? BLADERF_GAIN_AUTOMATIC : BLADERF_GAIN_MANUAL;
        status = bladerf_set_gain_mode(dev, BLADERF_MODULE_RX, gain_mode);

installation succesfull 

please correct if i wrong 

thanks

rtucker, 2018-02-07:
Whoops...

Could you try with the fix-issue-544 branch (https://github.com/Nuand/bladeRF/tree/fix-issue-544)?  I think this should fix it, but I want to make sure before I merge it down.

Thanks!  -rt

baguzajja, 2018-02-07:
i cant try to my device. because the device sill on the way to my home. 
i hope other can fix this problem

closed 2018-04-30 by rtucker