Nuand/bladeRF #553: Matlab: cannot re-open a bladeRF after closing it

rtucker, 2018-03-27:
Reported in the forums:

https://www.nuand.com/forums/viewtopic.php?p=7713#p7713

https://www.nuand.com/forums/viewtopic.php?p=7748#p7748

Steps to reproduce:

- Open a device: `device = bladeRF();`
- Delete the device using `clear device`
- Try to open the device again: `device = bladeRF();`
- Observe that the device cannot be found

Also observable by bladeRF-cli being unable to open the device while Matlab is still open.

rtucker, 2018-04-19:
Based on testing, it looks like #550 fixed this.  Holding this open until further confirmation occurs.

aricwang88, 2018-05-23:
I met the issue too, tx only work for once for this power up circle.

rtucker, 2018-05-23:
@aricwang88: is this with the above installer, or an earlier version?  if so, which version/installer?

aricwang88, 2018-05-30:
@rtucker , I built from the latest git, when I run srsue,  only can transmit for once for each power circle? But I found osmocom can work normally,  maybe some resource not released correctly by srsue.  I modified the srsenb and srsue calibration code,  still no progress.   srsLTE lack these part code to calibration DC and IQ imbalance.

robertghilduta, 2018-08-23:
@aricwang88 can you please give the MATLAB bindings (in master) or the latest windows installer a try? Please also ensure you're using the libusb driver.

rtucker, 2018-08-29:
Closing this issue, as the original issue relating to Matlab on Windows is fixed and any issues with srsLTE are unrelated to it.  Please open a new issue for any further issues.  Thanks :)

closed 2018-08-29 by rtucker