Nuand/bladeRF #554: Build breaking in Ubuntu 18.04 (pthreads vs pthread)

anshulthakur, 2018-04-29:
Hi, I'm trying to build bladeRF on a fresh Ubuntu 18.04 installation. Mine is an x86_64 machine. The build is failing as one of the cmake dependencies checks fails to find 'lpthreads' even though it is set in the CMake that the machine has 'pthread' and not 'pthreads'. Here is the truncated output of CMake Cache and Error:

```
//Dependencies for the target
libbladerf_shared_LIB_DEPENDS:STATIC=general;-lpthread;general;/usr/lib/x86_64-linux-gnu/libusb-1.0.so;
...
CMAKE_GENERATOR_TOOLSET:INTERNAL=
//Have symbol pthread_create
CMAKE_HAVE_LIBC_CREATE:INTERNAL=
//Have library pthreads
CMAKE_HAVE_PTHREADS_CREATE:INTERNAL=
//Have library pthread
CMAKE_HAVE_PTHREAD_CREATE:INTERNAL=1
//Have include pthread.h
CMAKE_HAVE_PTHREAD_H:INTERNAL=1
...
PKGCONFIG_LIBUSB_STATIC_LDFLAGS:INTERNAL=-lusb-1.0;-ludev;-pthread
PKGCONFIG_LIBUSB_STATIC_LDFLAGS_OTHER:INTERNAL=-pthread

```

```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/craft/gnuradio/default/src/bladeRF/host/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_e6de4/fast"
/usr/bin/make -f CMakeFiles/cmTC_e6de4.dir/build.make CMakeFiles/cmTC_e6de4.dir/build
make[1]: Entering directory '/home/craft/gnuradio/default/src/bladeRF/host/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_e6de4.dir/CheckSymbolExists.c.o
/usr/bin/cc  -I/usr/include/libusb-1.0  -fvisibility=hidden    -o CMakeFiles/cmTC_e6de4.dir/CheckSymbolExists.c.o   -c /home/craft/gnuradio/default/src/bladeRF/host/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_e6de4
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_e6de4.dir/link.txt --verbose=1
/usr/bin/cc  -fvisibility=hidden     -rdynamic CMakeFiles/cmTC_e6de4.dir/CheckSymbolExists.c.o  -o cmTC_e6de4 
CMakeFiles/cmTC_e6de4.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_e6de4.dir/build.make:97: recipe for target 'cmTC_e6de4' failed
make[1]: *** [cmTC_e6de4] Error 1
make[1]: Leaving directory '/home/craft/gnuradio/default/src/bladeRF/host/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_e6de4/fast' failed
make: *** [cmTC_e6de4/fast] Error 2

File /home/craft/gnuradio/default/src/bladeRF/host/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/craft/gnuradio/default/src/bladeRF/host/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_6a9f8/fast"
/usr/bin/make -f CMakeFiles/cmTC_6a9f8.dir/build.make CMakeFiles/cmTC_6a9f8.dir/build
make[1]: Entering directory '/home/craft/gnuradio/default/src/bladeRF/host/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_6a9f8.dir/CheckFunctionExists.c.o
/usr/bin/cc   -fvisibility=hidden -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_6a9f8.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.10/Modules/CheckFunctionExists.c
Linking C executable cmTC_6a9f8
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_6a9f8.dir/link.txt --verbose=1
/usr/bin/cc  -fvisibility=hidden -DCHECK_FUNCTION_EXISTS=pthread_create    -rdynamic CMakeFiles/cmTC_6a9f8.dir/CheckFunctionExists.c.o  -o cmTC_6a9f8 -lpthreads 
/usr/bin/x86_64-linux-gnu-ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_6a9f8.dir/build.make:97: recipe for target 'cmTC_6a9f8' failed
make[1]: *** [cmTC_6a9f8] Error 1
make[1]: Leaving directory '/home/craft/gnuradio/default/src/bladeRF/host/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_6a9f8/fast' failed
make: *** [cmTC_6a9f8/fast] Error 2
```

rtucker, 2018-04-30:
Hmm.  I can't reproduce this with a fresh install of Ubuntu 18.04 (from ubuntu-18.04-desktop-amd64.iso).  I've installed the prerequisites via `sudo apt-get install libusb-1.0-0-dev libusb-1.0-0 build-essential cmake libncurses5-dev libtecla1 libtecla-dev pkg-config git wget`.

My relevant chunks of CMakeCache.txt are identical, as far as I can tell:

```
//Dependencies for the target
libbladerf_shared_LIB_DEPENDS:STATIC=general;-lpthread;general;/usr/lib/x86_64-linux-gnu/libusb-1.0.so;
...
//Have symbol pthread_create
CMAKE_HAVE_LIBC_CREATE:INTERNAL=
//Have library pthreads
CMAKE_HAVE_PTHREADS_CREATE:INTERNAL=
//Have library pthread
CMAKE_HAVE_PTHREAD_CREATE:INTERNAL=1
//Have include pthread.h
CMAKE_HAVE_PTHREAD_H:INTERNAL=1
...
PKGCONFIG_LIBUSB_STATIC_LDFLAGS:INTERNAL=-lusb-1.0;-ludev;-pthread
PKGCONFIG_LIBUSB_STATIC_LDFLAGS_OTHER:INTERNAL=-pthread
```

I also get a CMakeFiles/CMakeError.log with the same failure, which is expected.

Can you confirm that you have all of the prerequisites installed?  Also, if you haven't yet, can you try removing the 'build' directory, re-creating it, then re-running `cmake ..` and post the output?

Thanks!  -rt

anshulthakur, 2018-05-01:
You are right @rtucker . On doing a standalone build, the installation proceeds normally. It is in the specific case of using PyBombs (pybombs install bladeRF) that creates this issue.

Thanks!
~Anshul

closed 2018-05-01 by anshulthakur

Cr4sh, 2019-05-01:
Can you please provide any technical details on fixing this issue? On my system cmake fails with the same error.