Nuand/bladeRF #557: Peristent USB transfer errors (Linux)

DavidCNou, 2018-06-06:
Hello, I'm having difficulty with the BladeRF in TX mode where I'm getting libusb transfer errors often. I'm trying to follow the DC and IQ calibration procedures at https://github.com/Nuand/bladeRF/wiki/DC-offset-and-IQ-Imbalance-Correction#correcting-the-tx-module . I can see the TX signal using GNURadio and another SDR for awhile and can use the sliders on the osmocom_siggen display as described in the cal procedure. After between 5secs to 3 minutes I get a few "Got transfer error for buffer" errors before a timeout error "[bladeRF sink] bladerf_sync_tx error: Operation timed out" and "[bladeRF sink] Consecutive error limit hit. Shutting down." I have to kill the siggen and restart it to get back running. I've also experienced these errors with RX but don't have those errors logged. Here are my system details:

Centos 7 x86_64

> uname -a

`Linux g10010-lnx 3.10.0-693.21.1.el7.x86_64 #1 SMP Wed Mar 7 19:03:37 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

* libusb packages
```
libusbx-devel-1.0.21-1.el7.x86_64
libusbx-1.0.21-1.el7.x86_64
```

> bladeRF-cli -i
```
bladeRF> open

bladeRF> print

  RX Bandwidth:  28000000 Hz
  TX Bandwidth:  20000000 Hz

  RX Frequency: 2484000000 Hz
  TX Frequency: 2467999999 Hz

   AGC: Disabled  

  GPIO: 0x0000002f
    LMS Enable:         Enabled   
    LMS RX Enable:      Enabled   
    LMS TX Enable:      Enabled   
    TX Band:            High Band (1.5GHz - 3.8GHz)
    RX Band:            High Band (1.5GHz - 3.8GHz)
    RX Source:          LMS6002D

  Loopback mode: none

  RX mux: BASEBAND_LMS - Baseband samples from LMS6002D

  RXLNA Gain:    6 dB
  RXVGA1 Gain:  30 dB
  RXVGA2 Gain:   3 dB
  TXVGA1 Gain:  -4 dB
  TXVGA2 Gain:  25 dB

  Sampling: Internal

  RX sample rate: 1000000 0/1
  TX sample rate: 20000000 0/1

  SMB Mode:   Disabled

  Current VCTCXO trim: 0x9581
  Stored VCTCXO trim:  0x9581

  VCTCXO tamer mode: Disabled

  Expansion GPIO register: 0xffffffff
  Expansion GPIO direction register: 0x00000000

bladeRF> probe

  Backend:        libusb
  Serial:         b90d3e2475556ec7b2ddfe4e12355349
  USB Bus:        4
  USB Address:    60

bladeRF> info

  Serial #:                 b90d3e2475556ec7b2ddfe4e12355349
  VCTCXO DAC calibration:   0x9581
  FPGA size:                40 KLE
  FPGA loaded:              yes
  USB bus:                  4
  USB address:              60
  USB speed:                SuperSpeed
  Backend:                  libusb
  Instance:                 0

bladeRF> version

  bladeRF-cli version:        1.5.1-git-23c6379
  libbladeRF version:         1.9.0-git-23c6379

  Firmware version:           1.8.0
  FPGA version:               0.7.1
```
* osmocom-siggen
```
LIBUSB_DEBUG=4 osmocom_siggen -s 20M -x 1M -f 2468M --sine --amplitude=1.0 -a verbosity=debug,buffers=128 > ~/dbg_out 2>&

[143.526640] [00000b28] libusb: debug [handle_events] poll() returned 1
[143.526669] [00000b28] libusb: debug [reap_for_handle] urb type=3 status=-71 transferred=5120
[143.526677] [00000b28] libusb: debug [handle_bulk_completion] handling completion status -71 of bulk urb 1/1
[143.526685] [00000b28] libusb: debug [handle_bulk_completion] low level error -71
[143.526692] [00000b28] libusb: debug [arm_timerfd_for_next_timeout] next timeout originally 3000ms
[143.526703] [00000b28] libusb: debug [usbi_handle_transfer_completion] transfer 0x7ff5a00818b0 has callback 0x7ff5c7c3d9c0
[ERROR @ "/home/username/bladeRF/host/libraries/libbladeRF/src/backend/usb/libusb.c":1005] Got transfer error for buffer 0x7ff5a0020fc0
[143.526716] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a00811b0
[143.530129] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a0081290
[143.533623] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a0081370
[143.533664] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a0081450
[143.537126] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a0081530
[143.537166] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a0081610
[143.540629] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a00816f0
[143.540670] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a00817d0
[143.544127] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a0081990
[143.544142] [00000b28] libusb: debug [discard_urbs] URB not found --> assuming ready to be reaped
[143.544148] [00000b28] libusb: debug [libusb_cancel_transfer] cancel transfer failed error -5
[143.544160] [00000b28] libusb: debug [libusb_cancel_transfer] transfer 0x7ff5a0081a70
```
`reap_for_handle` is in libusb/os/linux_usbfs.c:2643 and is looking at `urb->status` 
Low level errno -71 is -EPROTO . My kernel's usb documentation has this to say about -EPROTO:
```
**************************************************************************
*                   Error codes returned by in urb->status               *
*                   or in iso_frame_desc[n].status (for ISO)             *
**************************************************************************

USB device drivers may only test urb status values in completion handlers.
This is because otherwise there would be a race between HCDs updating
these values on one CPU, and device drivers testing them on another CPU.

A transfer's actual_length may be positive even when an error has been
reported.  That's because transfers often involve several packets, so that
one or more packets could finish before an error stops further endpoint I/O.

For isochronous URBs, the urb status value is non-zero only if the URB is
unlinked, the device is removed, the host controller is disabled, or the total
transferred length is less than the requested length and the URB_SHORT_NOT_OK
flag is set.  Completion handlers for isochronous URBs should only see
urb->status set to zero, -ENOENT, -ECONNRESET, -ESHUTDOWN, or -EREMOTEIO.
Individual frame descriptor status fields may report more status codes.

[snip]

-EPROTO (*, **)         a) bitstuff error
                        b) no response packet received within the
                           prescribed bus turn-around time
                        c) unknown USB error
[snip]
(*) Error codes like -EPROTO, -EILSEQ and -EOVERFLOW normally indicate
hardware problems such as bad devices (including firmware) or cables.

(**) This is also one of several codes that different kinds of host
controller use to indicate a transfer has failed because of device
disconnect.  In the interval before the hub driver starts disconnect
processing, devices may receive such fault reports for every request.
```

Without digging into the BladeRF USB firmware, I'm stumped. I've got the most updated libusb I believe and the newest BladeRF driver and firmware loaded (at least as of a week ago). The only thing is my kernel is quite old due to Centos 7. Can any developer please take a look as to what might be causing the problem? Perhaps something with USB power saving or auto suspend (I think I've disabled it in `/etc/default/tlp` with `USB_BLACKLIST="1d50:6089 1d50:6066"`)

Thanks
Dave C

robertghilduta, 2018-08-23:
@DavidCNou can you please give libbladeRF 2.x (the lastest master) a try? We had a number of sample buffering fixes go in that might positively impact the issue you're seeing.

closed 2018-09-12 by rtucker

DavidCNou, 2018-09-14:
Can you please explain why this issue was closed? Were you able to reproduce and determine a cause of the errors?

rtucker, 2018-09-14:
A response was sent on August 23 requesting a re-test with the latest libbladeRF, and no response was received, so the issue was closed in the interim.  If you could try to reproduce with libbladeRF 2.0.2 or later, firmware 2.2.0 or later, and FPGA 0.7.3 or later, it would be most appreciated.  Feel free to re-open this issue if the problem still occurs and we'll take another look.

Thanks!

DavidCNou, 2018-09-18:
I will give it a try when I am back near the hardware. I've been on travel for the past 3 weeks and am unable to re-test until I return to the office. 

By the way, I waited 6 weeks to hear any response from my issue and in the mean time I used a HackRF instead to complete my public demo.

rtucker, 2018-09-20:
I apologize for the delay in the response.  In general, issues opened via GitHub are handled as part of our development process, and we can't guarantee if or when a particular issue will be addressed.

For future reference, we do provide support for customer issues via forums, Slack, IRC, and e-mail; see https://www.nuand.com/support/ for further information.

TurboDirectInjection, 2021-06-28:
I experienced the same issue as you, and for me what fixed it was rolling back to an earlier FPGA image.  It was, apparently, the combination of Ubuntu 20.04 and FPGA v0.12.0 for BladeRF x40 that was the problem.  Relevant thread [here](https://www.gitmemory.com/issue/Nuand/bladeRF/819/866455237).