Nuand/bladeRF #558: Apparent dropped samples during rx sample rate test

brianmwaters, 2018-06-15:
I followed the rx sample rate test at https://github.com/Nuand/bladeRF/wiki/Debugging-dropped-samples-and-identifying-achievable-sample-rates#looking-for-gaps-in-received-samples-using-an-fpga-counter, and found that at high sample rates (over ~39.5 MS/s) I see dropped samples about 80% of the time, even on different hardware.

This might seem normal, but I've noticed that the gap _always_ starts at 15360 samples, or 61440 bytes, into the stream, which is 4096 bytes short of 2^16.

I haven't bothered to do a USB capture to figure out if this is a host-side or a device-side issue (sorry), but I thought I'd leave it here for someone more familiar with the bladeRF platform to have a look at.

robertghilduta, 2018-08-23:
@brianmwaters this was fixed in the latest libbladeRF version bump, please give master a try and let us know if the problem persists.

closed 2018-09-12 by rtucker

brianmwaters, 2018-09-26:
I finally got around to testing this, and the issue does indeed seem fixed. Thanks!

I'm curious what the issue was. Can you point me at any particular commits?

rtucker, 2018-09-26:
Looking back over the commits and my journal, I can't remember exactly when it was fixed, or which problem it was, but chasing down a "4096 bytes short of 2^16" gap sounds very familiar.

I looked through the commit log, but yeah... on just the FPGA side, the FIFOs got reworked, the FX3 GPIF controller got redesigned, and we added more flow control and pipelining, and any of those could have done the trick.

Sorry I can't be more specific!  I'm glad it actually fixed your issue, though :)