Nuand/bladeRF #564: checksum files missing from download page for latest firmware revision

sparklespdx, 2018-08-13:
Hey folks,

I noticed when I went to download the firmware for the A4 that the checksum files for revision v0.7.3 and v0.7.1 throw a 404 when I try to visit them. It looks like the links in the v0.7.3 section actually point to the v0.7.1 checksums, but the download link looks OK.

https://www.nuand.com/fpga/v0.7.1/hostedxA4.rbf.sha256sum

[image]

It was great stopping by your booth at Def Con, really looking forward to using the radio!

robertghilduta, 2018-08-14:
The links have been updated. Thanks for pointing this out!

sparklespdx, 2018-08-14:
:+1: you bet, thanks for the quick response!

closed 2018-08-14 by sparklespdx