Nuand/bladeRF #586: bladeRF-cli: Use readline and/or editline in interactive mode

dlrobertson, 2018-08-24:
If [editline] or [readline] are available, one of them could be used to add more features to interactive mode (e.g. history).

[editline]: https://linux.die.net/man/3/editline
[readline]: https://linux.die.net/man/3/readline

dlrobertson, 2018-08-24:
I am interested in working on this if this is something that would be useful. I have briefly looked through [bladeRF-cli/src/input](https://github.com/Nuand/bladeRF/tree/master/host/utilities/bladeRF-cli/src/input), but any other input would be appreciated.

bglod, 2018-08-24:
If you have libtecla installed on your system prior to building bladeRF, it will provide command history and tab completion in the CLI. Have you tried that first?

dlrobertson, 2018-08-24:
Ah! It seems the Arch Linux bladerf PKGBUILD doesn't currently support libtecla. So 1) The PKGBUILD should probably be updated (I can whip up a patch to the bladerf PKGBUILD as libtecla is also a community supported PKGBUILD) 2) Would it be helpful to support multiple backends (readline, editline, and tecla)? so that if libtecla isn't installed, but readline is you still have history etc.

bglod, 2018-08-24:
A fellow Arch user! Thanks for letting me know that libtecla got accepted into community -- I had been running it from the AUR forever.

As for multiple backends, it would certainly be neat if it's not too much trouble to implement. How much effort do you think it would be? My other concern is future support issues that may arise (e.g. something works for libtecla but not for editline/readline). If you think these concerns are minimal and you feel up to it, I would gladly take a look at any PR you submit.

dlrobertson, 2018-08-24:
> A fellow Arch user! 

Lol yup! Submitted the patch to the community packages issue list.

> As for multiple backends, it would certainly be neat if it's not too much trouble to implement. How much effort do you think it would be?

Awesome! I'm not entirely sure at the moment. I need to do a bit more research (I only just got my bladerf :smile:), but typically adding readline/editline support is pretty easy. I would expect it to be close to `tecla.c`.

> My other concern is future support issues that may arise (e.g. something works for libtecla but not for editline/readline)

Yeah this is definitely something to consider.

rtucker, 2018-08-24:
G'day,

Only concern I'd have is GNU Readline is a GPL library (rather than LGPL), which would complicate licensing stuff.  But NetBSD libedit should be fine, since it's BSD-licensed.  I would quite enjoy a libedit option, though.

If you haven't yet, we do have an Individual Contributor Assignment Agreement which we require for significant contributions: https://github.com/Nuand/bladeRF/tree/master/legal/CAA

tl;dr: this would be most awesome, go with libedit :)

dlrobertson, 2018-08-27:
Just realized this is a duplicate of https://github.com/Nuand/bladeRF/issues/319

closed 2018-08-27 by dlrobertson