Nuand/bladeRF #590: fpga hosted image synthesis failed at "Building BSP and sample application..."

Gigithecode, 2018-08-27:
Hello,

On Ubuntu 16.04 - Gnu/Linux
gigi@clevo:~/bladeRF/hdl/quartus/work/bladerf-115-hosted$ uname -a
Linux clevo.gigi.edu 4.13.0-46-generic #51-Ubuntu SMP Tue Jun 12 12:36:29 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

What ever I tried, I cannot succeed to synthesis fpga image.
The git branch is 'origin/master'
With Quartus 16 web edition (lite) - I tried from 13.0 to 18.0

./build_bladerf.sh -r hosted -s 115 -b bladeRF
...
##########################################################################
    Building BSP and bladeRF application...
##########################################################################

boost libraries not loaded
channel type customly must define watchProc
Aborted (core dumped)

gigi@clevo:~/bladeRF/hdl/quartus/work/bladerf-115-hosted$ core
-rw------- 1 gigi gigi 1198780416 août  27 18:47 core
[New LWP 13985]
[New LWP 13965]
[New LWP 13966]
[New LWP 13969]
[New LWP 13968]
[New LWP 13972]
[New LWP 13970]
[New LWP 13971]
[New LWP 13975]
[New LWP 13974]
[New LWP 13973]
[New LWP 13976]
[New LWP 13978]
[New LWP 13964]
[New LWP 13967]
[New LWP 13977]
[New LWP 13982]
[New LWP 13979]
[New LWP 13983]
[New LWP 13980]
[New LWP 13984]
[New LWP 13981]
Core was generated by `/home/gigi/intelFPGA_lite/16.1/quartus/linux64/jre64/bin/java -cp /home/gigi/in'.
Program terminated with signal SIGABRT, Aborted.
#0  0x000014c65b6990bb in ?? ()
[Current thread is 1 (LWP 13985)]
(gdb) #0  0x000014c65b6990bb in ?? ()
#1  0x0000000000000004 in ?? ()
#2  0x000014c632b97d90 in ?? ()
#3  0x0000000000000000 in ?? ()
(gdb) quit
\n

gigi@clevo:~/bladeRF/hdl/quartus$ ~/utility/build/pargs work/bladerf-115-hosted/core
core 'work/bladerf-115-hosted/core' of 13964: /home/gigi/intelFPGA_lite/16.1/quartus/linux64/jre64/bin/java -cp /home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/TableLayout.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/appframework-1.03.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/beansbinding-1.2.1.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/binding-2.0.6.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.app_lib.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.bsp.editor.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.bsp.elements.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.bsp.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.bsp.plugin.core.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.bsp.plugin.hal.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.bsp.plugin.lwhal.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.bsp.schema.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.bsp.scripting.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.commandline.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.cpexample.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.flash.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.launch.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.swinfo.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.embeddedsw.utilities.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.entityinterfaces.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.hdlcomponent.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.hdlwriter.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.infrastructure.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.jdbcsqlite.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.jtagsimulator.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.librarian.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.megawizard2.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.minieval2.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.nios2.isa.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.nios2.rtl.trace.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.nios2.trace.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.privateinterfaces.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.qsys.blackboxmodule.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.qsys.cmsis.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.qsys.ipxact.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.qsys.ipxact.module.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.qsys.model.common.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.service.jre.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopc.generator.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopc.qsymbol.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcdocument.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcfactories.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopclibrary.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcmodel.atlantic.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcmodel.components.atlantic.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcmodel.components.tclmodule.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcmodel.embeddedsw.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcmodel.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcmodel.transforms.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcplatform.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcreport.build.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.sopcreport.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.systemconsole.configrom.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.systemconsole.designs.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.systemconsole.driver.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.systemconsole.gui.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.systemconsole.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.systemconsole.lightclient.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.systemconsole.plugin.elf.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.systemconsole.plugin.jtag.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.systemconsole.scripting.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.tcl.interpreter.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.tcl.ui.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.tools.exampledesigns.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.ui.ientity.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.ui.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.ui.quartus.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.util.elf.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.utilities.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/com.altera.version.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/commons-lang3-3.1.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/commons-logging-1.1.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/docking-frames-common.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/docking-frames-core.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/explicitlayout.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/guava-15.0.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/hamcrest-all-1.3.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/jacl1.3.2a.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/javasysmon.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/jaxb-api.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/jaxb-core.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/jaxb-impl.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/jline-2.12.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/jsr173_1.0_api.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/miglayout15-swing.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/mydoggy-api.jar:/home/gigi/intelFPGA_lite/16.1/nios2eds/sdk2/lib/ [... cut ...]
argv[20]: hal.enable_instruction_related_exceptions_api
argv[21]: false
argv[22]: --set
argv[23]: hal.make.bsp_cflags_optimization
argv[24]: -Os
argv[25]: --set
argv[26]: hal.enable_exit
argv[27]: 0
argv[28]: --set
argv[29]: hal.enable_small_c_library
argv[30]: 1
argv[31]: --set
argv[32]: hal.enable_clean_exit
argv[33]: 0
argv[34]: --set
argv[35]: hal.enable_c_plus_plus
argv[36]: 0
argv[37]: --set
argv[38]: hal.enable_reduced_device_drivers
argv[39]: 1
argv[40]: --set
argv[41]: hal.enable_lightweight_device_driver_api
argv[42]: 1

bglod, 2018-08-27:
I can't seem to reproduce this (un)fortunately.

1. Have FPGA builds worked for you in the past?
2. Have you tried with a clean ```work``` directory?

```channel type customly must define watchProc``` seems to be the cause of the segfault. This is a Tcl or Quartus thing, so there might be an issue with your libraries or dependencies?

Our main build machine is Ubuntu 16.04, and I was able to build master just now using the following versions of Quartus:

1. Quartus Standard v16.0.1
2. Quartus Standard v17.1
3. Quartus Standard v18.0
4. Quartus Lite v18.0

We also have a build bot that runs through the permutations of lite and standard Quartus. If I remember correctly, the built bot starts at v16.0 and goes to the latest v18.0. I tend to recommend using the latest version of Quartus. While 16.0 works now, it's over 2 years old and I don't know how long it will continue to work with our code base as we move forward.

rtucker, 2018-08-27:
I don't recall exactly what happens if I don't, but on my Quartus installs, I usually delete three of the included library files:

```
${QUARTUS_PATH}/quartus/linux64/libboost_system.so
${QUARTUS_PATH}/quartus/linux64/libccl_curl_drl.so
${QUARTUS_PATH}/quartus/linux64/libstdc++.so.6
```

(ref: https://github.com/rtucker/bladeRF-buildbot/blob/dev-modernize/docker-quartus/Dockerfile)

This may break your Quartus install entirely, but it does work on my Ubuntu 16.04 environment.

Gigithecode, 2018-08-27:
Hello,
Yes, I succeeded to build it in the past (probably with Quartus web 15)
Yes, I tried removing work but it's doesn't help...

Part of my env after running ~/intelFPGA_lite/18.0/nios2eds/nios2_command_shell.sh ;

QUARTUS_ROOTDIR=/home/gigi/intelFPGA_lite/18.0/quartus
SOPC_KIT_NIOS2=/home/gigi/intelFPGA_lite/18.0/nios2eds
PATH=/home/gigi/.local/bin:/home/gigi/.nvm/versions/node/v10.9.0/bin:/home/gigi/intelFPGA_lite/18.0/nios2eds/bin/gnu/H-x86_64-pc-linux-gnu/bin:/home/gigi/intelFPGA_lite/18.0/nios2eds/sdk2/bin:/home/gigi/intelFPGA_lite/18.0/nios2eds/bin:/home/gigi/intelFPGA_lite/18.0/modelsim_ase/linuxaloem:/home/gigi/intelFPGA_lite/18.0/quartus/bin:/home/gigi/intelFPGA_lite/18.0/quartus/sopc_builder/bin...

JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64/

The crash occurs in a java process "com.altera.embeddedsw.commandline.bsp.BspCreateSettingsCommand"

So I was wondering if something in my java configuration was messy... but Altera uses it's own embedded java VM "/home/gigi/intelFPGA_lite/16.1/quartus/linux64/jre64/bin/java"

Whis is ;
java version "1.8.0_05"
Java(TM) SE Runtime Environment (build 1.8.0_05-b13)
Java HotSpot(TM) 64-Bit Server VM (build 25.5-b02, mixed mode)

I cannot figure out what's the problem.
I'm going to try in a chrooted env. in case something mess in my configuration.

Gigithecode, 2018-08-27:
I deleted libssl, libcrypto and liblzma.so from quartus/linux64 and link my system lib.
I'm going to try to delete libboost_system.so, libccl_curl_drl.so and libstdc++.so.6 as well...

Gigithecode, 2018-08-27:
Ok, i cloned bladeRF-buildbot, 
I've got a docker up and running...
How to build and run the image?

rtucker, 2018-08-28:
Basically,

1. Get the `dev-modernize` branch of bladeRF-buildbot
2. `cd docker-quartus` and `./do_build`
3. Wait for it to download and install ~10 GB of Quartus Prime
4. Check out the bladeRF repository somewhere and cd to the base of it (`git clone https://github.com/Nuand/bladeRF.git && cd bladeRF`)
5. `docker run --rm -i -t -v /sys:/sys:ro -v $(pwd):/build quartus-lite`
6. You will find yourself in the `Altera Nios2 Command Shell`
7. `cd /build/hdl/quartus`
8. `./build_bladerf.sh -c -b bladeRF -r hosted -s 115` (or other arguments as desired)

I believe that will do it!  If it does, you might also want to compare the list of packages installed by that Dockerfile with what's installed on your system... it's possible some dependency is missing somewhere.

Gigithecode, 2018-08-29:
Thank you a lot ! I'll have a look on it !

Gigithecode, 2018-09-03:
It works well! Thank you...
I'll analyze why the "native" compilation failed...

closed 2018-09-05 by rtucker

kempd1, 2023-10-20:
I'm running into the same issue running Quartus 17.1 with the native compilation, seems like the docker file for bladeRF-buildbot, was broken somewhere between Ubuntu 16 and Ubuntu 22.  Did we figure out why  the ```boost libraries not loaded
channel type customly must define watchProc Aborted (core dumped)``` issue was occurring?